artisan-roaster-scope/artisan #401: Batches and Ranking Reports; Web and Excel formats

DennieSch, 2019-10-09:
## Expected Behavior
Nominal output data and formatting. 

## Actual Behavior
Some anomalies and inconsistencies; (friendly) details provided in the attached.
[Report Comments (v2.0.1).zip](https://github.com/artisan-roaster-scope/artisan/files/3709474/Report.Comments.v2.0.1.zip)

Using version = 2.0.1 (5ffc787b)

## Steps to Reproduce the Problem

  1. Standard report generating steps. 
  2. 
  3. 

## Specifications

version = 2.0.1 (5ffc787b)
Architecture = ('64bit', 'WindowsPE')
Machine = AMD64
Platform name = Windows-10-10.0.18362-SP0
Processor = Intel64 Family 6 Model 58 Stepping 9, GenuineIntel
Python Branch = tags/v3.7.4
Python Build = ('tags/v3.7.4:e09359112e', 'Jul 8 2019 20:34:20')
Python Compiler = MSC v.1916 64 bit (AMD64)
Python Implementation = CPython
Python Revision = e09359112e
Python version = 3.7.4
Release = 10
System = Windows
Version = 10.0.18362
Win32 = ('10', '10.0.18362', 'SP0', '')

  - Connected devices or roasting machine: N/A
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2019-10-10:
Thanks for this detailed report. I will try to fix the reported issues. It would help if you could provide a set of sample .alog files that demonstrate the issue with the wrong LOSS average computation (on slide 3). The issue might be in the profiles not the report as I could not duplicate this one. Thanks!

DennieSch, 2019-10-10:
Hey Marko- That was guick....you 'da man!
Here are the files (.alog, .aset, .html). 

Cheers- Dennie
[Profiles and Settings file.zip](https://github.com/artisan-roaster-scope/artisan/files/3713670/Profiles.and.Settings.file.zip)

MAKOMO, 2019-10-11:
Thanks for those files. I think I resolved everything by now, thus I close this one.

closed 2019-10-11 by MAKOMO

DennieSch, 2019-10-14:
Hey Marko- 
Looks like the Roast/Ranking/Excel... report still has "%" marks in the data fields. 
[percent marks in data.zip](https://github.com/artisan-roaster-scope/artisan/files/3722975/percent.marks.in.data.zip)

MAKOMO, 2019-10-14:
Excel is doing this as the cell format is set to percent. Nothing to do about it. I think it is fine here. The cell itself contains only the number (0.21 rendered as 21% by Excel).

DennieSch, 2019-10-14:
I do understand what you are saying and the cell format you mention. The report I generated with version = 2.0.1 (cd0a22e9) does have the symbol in the cell. Take a peek at the attached when you have a chance and see what ya think.
[ranking2.zip](https://github.com/artisan-roaster-scope/artisan/files/3723052/ranking2.zip)

Cheers - Dennie

MAKOMO, 2019-10-14:
I am puzzled now. I took your file and set the cell formatting to number and did not see any % symbol. They are pure numbers. Only the percent formatting is adding the %. See files attached. Maybe it should be removed from the header here to avoid the duplication. I will do that.

[ranking2-number-format.zip](https://github.com/artisan-roaster-scope/artisan/files/3723232/ranking2-number-format.zip)

DennieSch, 2019-10-14:
Hi Marko- I see what you mean. I should have caught that before sending the comment...'my  bad'.

Here's what I was thinking....there could be two approaches-- 

1) % marks in the data and not in the header row
-- application passes the data to excel as a decimal (0.XXXXXXXX) and the cell is formatted as a percentage
-- Result is "XX.X%" for the data in the cells

2) % marks in the header row and not in the data
-- application passes the data (assumed to be carried as a decimal) as 'DATA*100' to excel and the cell is formatted as a number
-- Result is "XX.X" for the data in the cells
-- any 'SUM' or 'AVG' rows would need to be dealt with accordingly

Do you think #2 is doable?? I'm thinking that would make for 'cleaner' and consistent looking reports.

Cheers,
Dennie

MAKOMO, 2019-10-18:
Dear Dennie,

 I decided for approach 1 here as the Excel export is not about "clean look", but about data that can be easily further processed. Sorry.

DennieSch, 2019-10-18:
Hi Marko- Thanks for considering this and your reply. While I might have been leaning more towards approach 2, I respect your decision and support 100%. No problem.
Cheers- Dennie