artisan-roaster-scope/artisan #405: Device Assignment window; size, resizing, position

DennieSch, 2019-10-14:
## Expected Behavior
Window rendered within the screen on initial opening and retain size and position after resizing.

## Actual Behavior
Doesn't seem to respect right side of display on opening (goes off screen) nor position
(see Initial)

## Steps to Reproduce the Problem

  1. Move window to the left to be able to resize the window. Move right side of window to the left the maximum amount possible.
(see resized window). 
  2. Click OK.
  3. Reopen window and it is back to the original size (with the left side off screen).
(see reopened window)

Resize width of window to minimum or any other size so the entire window is visible. Click Ok. Reopen window. The application seems to remember the position of the left side of the window but the right size is again off the display screen. 

Alarms window
Seems to be the only window with the Minimize Maximize/Restore window buttons on the top right corner of the window....which works well. Maybe Device Assignment window and other windows (where it makes sense...variable/long data field entries) should have this attribute.

## Specifications
[Device Assignment window.zip](https://github.com/artisan-roaster-scope/artisan/files/3722942/Device.Assignment.window.zip)

 version = 2.0.1 (cd0a22e9)
Architecture = ('64bit', 'WindowsPE')
Machine = AMD64
Platform name = Windows-10-10.0.18362-SP0
Processor = Intel64 Family 6 Model 58 Stepping 9, GenuineIntel
Python Branch = tags/v3.7.4
Python Build = ('tags/v3.7.4:e09359112e', 'Jul 8 2019 20:34:20')
Python Compiler = MSC v.1916 64 bit (AMD64)
Python Implementation = CPython
Python Revision = e09359112e
Python version = 3.7.4
Release = 10
System = Windows
Version = 10.0.18362
Win32 = ('10', '10.0.18362', 'SP0', '')
  - Connected devices or roasting machine: none
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2019-10-23:
I cannot reproduce this on my Mac, but I made a change (Commit 4e49b202d47017d8e234dbfa4dd73506b6ca4e3b) that could improve the situation. Could you test this one and see if it resolve the issue?

I added also the min/max buttons for Windows to the following dialogs
- roast properties
- background
- cupping
- device
- port
- events

DennieSch, 2019-10-24:
Hi Marko-
This might be a little lengthy but hopefully it is of some help.  Here it goes......

Running version = 2.0.1 (88193cf3)

Not sure we're there yet. 
Device Assignment window
It appears this window has a default size it likes to open to.
- If I resize the width smaller or to the smallest, click OK, and then reopen... the window is back to its 'default size' with the left edge where I left it and the right edge moved out to the right (either on screen or off screen depending on where I had placed the left edge.

If I resize the width to larger than its 'default size', click OK and then reopen... the larger window is the size I had left it (resized large) and in the position I left it. 

After writing the above, thought I would try and go through all/most of the windows.

Config Menu.....

Roast Profile
- remembers resize and position

Background
- reverts back to a default size if resized to smaller than its 'default'
- remembers position except for when it resizes itself (by extending the right side)
- no Cancel button

Cup Profile
- remembers resize and position
- no Cancel button

Device Assignment
- reverts back to a default size if resized to smaller than its 'default'. 'Default' size seems to be much larger (wide) than it probably needs to be.
- remembers position except for when it resizes itself (by extending the right side)

Ports Configuration
- always resizes itself to a default size and position

Sampling Interval
- a resizing handle does appear when hovering on the left-side edge (only) but doesn't do anything (don't believe there is a need to resize this window)
- default window size is too small; it prevents the title of the window "Sampling Interval" to be displayed in it entirety.
- after repositioning, always reopens in its default position

Curves
- no resizing handles
- after repositioning, always reopens in its default position

Events
- reverts back to a default size if resized to smaller than its 'default'
- remembers resize and position

Alarms
- no Cancel button
- remembers resize and position

Phases
- displays nominally
- no resizing handles; which is okay
- after repositioning, always reopens in its default position

Statistics
- no Cancel button
- displays nominally
- no resizing handles; which is okay
- after repositioning, always reopens in its default position

Axes, Colors, Autosave and Batch
- a resizing handle does appear when hovering on the left-side edge (only) but doesn't do anything (don't believe there is a need to resize these windows)
- after repositioning, always reopens in its default position

---------------------------------

Tools Menu……
Calculator
- remembers resize and position
- RECOMMENDATION – add a Close button to bottom of window
- RECOMMENDATION – add a Calculate button to bottom of window

Wheel Graph – Editor
- always resizes itself to a default size and position
- default size should be increased to initially display all columns of the table

--------------------------

The Min/Max buttons are working good.

Hope this is of some help.

Cheers,
Dennie

MAKOMO, 2019-11-06:
Thanks for this list of the current state on different dialogs and the confirmation that the position of the Device Assignment window is now correctly remembered.

If time allows we will add Cancel buttons to more dialogs and make them remember their position. However, for now I think this is pressing for any of those or not even needed.

closed 2019-11-06 by MAKOMO