artisan-roaster-scope/artisan #407: Roast/Properties -- Color

DennieSch, 2019-10-15:
## Expected Behavior
Have the ability to enter and record Roast Level in traditional terms when one doesn't have access to an electronic or automated roast color meter.

## Actual Behavior (Suggested Behavior)
Add a selection like "Roast Level" to the drop-down currently containing a list of roast color meters.
When "Roast Level" is selected, the 'Whole' and 'Ground' data entry fields would not only accept numeric entries but also alpha and special characters. Examples of entries might be; Cinnamon, American, City, City+, Full City, Full City+, Vienna, French, Italian....or, Light, Medium, Medium Dark, Dark.
The entered roast level for whole and/or ground would be available in the other parts of the application in place of the 'color number' such as, Exports, Reports and Summary. Having this information in Exports and Reports enables search, sorting and filtering based on Roast Level. This would be especially helpful when one has a significant number of roasts to assess, review, compare and analyze. 

The above is a suggested implementation. Maybe there are other implementations which provides this capability and is more compatible with the application's architecture or future capabilities.

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications
version = 2.0.1 (a282d326)
Architecture = ('64bit', 'WindowsPE')
Machine = AMD64
Platform name = Windows-10-10.0.18362-SP0
Processor = Intel64 Family 6 Model 58 Stepping 9, GenuineIntel
Python Branch = tags/v3.7.4
Python Build = ('tags/v3.7.4:e09359112e', 'Jul 8 2019 20:34:20')
Python Compiler = MSC v.1916 64 bit (AMD64)
Python Implementation = CPython
Python Revision = e09359112e
Python version = 3.7.4
Release = 10
System = Windows
Version = 10.0.18362
Win32 = ('10', '10.0.18362', 'SP0', '')
  - Connected devices Ror roasting machine: none

Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2019-10-23:
Roast color names are not general agreed and different regions on the planet are using different vocabulary. In EU city/full-city is unknown. One would need to make those color names configurable. How to map those names to the different numeric roast color scales?

This might not get added in the near future, but I keep it open.

DennieSch, 2019-10-23:
I would have to agreed. In searching the net, I have come across a variety of roast color descriptions as well as comparisons to the Agtron scale. 

As you mentioned, one way would be to make the color names configurable. I had previously been using 'RoasterThing'. It provides for the user to define (and it retained in a table) a list of roast levels (with a check box to enable or disable). The enabled levels were then available in a drop-down in two areas; 1) for the green coffee (inventory) for which one could select a Target Roast level that would be 'remembered' for that coffee for future reference/roasts and 2) in the roast (like Roast Properties of Artisan) as a Actual Roast level. While nice, I really wasn't suggesting this approach. 

What I was thinking was to have an additional selection in the color meter drop-down like "Roast Level" or "User Defined" or something like that. When this option was selected, the text boxes for Color under Whole and Ground would allow alphanumeric entries (I believe they are currently limited to numeric entries for the color meter selected... and should remain that way). It's when this 'additional selection' is chosen the user can enter what ever roast level descriptor they'd like (with a character limit). If they want to enter Full City or FC+, they would be able to do so. And, just like weight, volume, etc. it would need to be entered each time one performs a roast. It would be up to the user to maintain consistency in their descriptors. 

So... anyway... that is what I was thing... for what it's worth :-)

Maybe something to consider for the future. 

Cheers,
Dennie

MAKOMO, 2020-05-19:
I rather not spam the app with this. A color name can well be added to the roasting notes.

closed 2020-05-19 by MAKOMO

DennieSch, 2020-05-19:
That is what I've been doing; adding a color name to the notes. 
In addition, I've also created my own numbering system based on color. Then, I enter that number into the Ground Color in roast properties. That way I can get a 'color representation' to come out in the various reports (e.g. excel). This allows me to do sorts, plots and correlations using excel to compare and analyze roasts.