artisan-roaster-scope/artisan #408: RoR and BT cuve stretched to the right after going out of Designer mode

dukeja, 2019-10-15:
## Expected Behavior
I was using the "Designer" mode to duplicate an existing roasting profile (opened as background).  I can design the BT curve and RoR perfect match the target profile.
After de-select the "Designer" (out of Designer mode), I should see a the same BT curve and RoR that I can save as profile.

## Actual Behavior
After going out of "Designer mode" the designed curve has similar shape, but the TP of BT is shifted slighted to the right for about 15 seconds.  The RoR at TP has large slope (almost vertical) and the template is also shifted to the right for about 15 seconds.  However, both Dry End and 1C point in the created template are stayed exactly the same before and after going out of "Designer Mode"

It is very annoying and actually created a wrong RoR curve that does not match to the previous existing profile.  The new designed template now actually misleads the user to a wrong profile curve and failed the roasting.

## Steps to Reproduce the Problem

  1. Open an successful roasting profile with BT and RoR
  2. Switch it to the background and Check the "Designer" to create a new template 
  3. Use "Config" and adding/moving data points to form a template BT curve perfectly mimicking the background BT curve (BTW, why the background RoR is turned to be invisible???)  You think that you have created a perfect template that mimic a previous successful roasting curve.
  4. Now turn off (leaving) the Designer mode, you will see the template BT curve now shifted to the right by about 15 seconds.  And you can see clearly the discrepancy between the designed template BT and RoR curve vs. the background BT and RoR curves.  The discrepancy is large at around TP and gets smaller close to the ending time.  And the Dry  End and 1C point are correctly matched, so that you cannot simply shifted the template (when used as the background) to correct this unintended shift since the shift amount is non-uniform.

It seems to be bug in the curve fitting algorithm after switching off the Designer mode.

## Specifications

  - Artisan Version: 1.6.2 
  - Artisan Build (number in brackets shown in the about box): 9d1202a4
  - Platform (Mac/Windows/Linux + OS version): Windows 10
  - Connected devices or roasting machine: Phidget 1101, Kaldi Wide roaster

[Roaster.zip](https://github.com/artisan-roaster-scope/artisan/files/3727615/Roaster.zip)

  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

roasterdave, 2019-10-15:
Thank you for the detailed report and for including your .aset and .alog files.  That made it possible to see what is happening.  Artisan is working as intended.  However, heavy curve smoothing was set under Config>Curves>Filters.  Smoothing causes time shifts in the curves.  To improve the alignment of your curves either reduce the amount of smoothing or check the option "Optimal Smoothing Post Roast". 

Further information can be found in the [Quick Start Guide](https://artisan-scope.org/docs/curves/) and the [Artisan Blog](https://artisan-roasterscope.blogspot.com/2014/01/sampling-interval-smoothing-and-rate-of.html)

dukeja, 2019-10-15:
OK, I got it.  I have tried to make the background curve with minimum filtering size.  Then design the curve and going out of it will not shift it.

I understand the limitation of traditional time invariant filtering, which will shift the filtered data.  I have been using wavelet-based filtering (pywavelet) and it has not such issue.  Actually I often export data out of Artisan and re-process it with my python code.   If you are interested, I may be able to show you how it perform.

Thanks!

roasterdave, 2019-10-20:
The 'Optimal Smoothing Post Roast' option applies smoothing without shifting the curve.   Does pywavelet provide a result that is meaningfully different?

I will close this as the original issue has been addressed.

closed 2019-10-20 by roasterdave