artisan-roaster-scope/artisan #410: Drop-down text box too small; Config/Events/Buttons - Button Size

DennieSch, 2019-10-20:
## Expected Behavior
Choices displayed in their entirety

## Actual Behavior
Only 1 to 2 characters are displayed (see attached)

## Steps to Reproduce the Problem

  1. Select: Config/Events/Buttons - Button Size
  2. 
  3. 

## Specifications

[Button Size - drop down.zip](https://github.com/artisan-roaster-scope/artisan/files/3747432/Button.Size.-.drop.down.zip)

version = 2.0.1 (e7606fd4)
Architecture = ('64bit', 'WindowsPE')
Machine = AMD64
Platform name = Windows-10-10.0.18362-SP0
Processor = Intel64 Family 6 Model 58 Stepping 9, GenuineIntel
Python Branch = tags/v3.7.4
Python Build = ('tags/v3.7.4:e09359112e', 'Jul 8 2019 20:34:20')
Python Compiler = MSC v.1916 64 bit (AMD64)
Python Implementation = CPython
Python Revision = e09359112e
Python version = 3.7.4
Release = 10
System = Windows
Version = 10.0.18362
Win32 = ('10', '10.0.18362', 'SP0', '')

  - Connected devices or roasting machine:none
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2019-10-23:
fixed

closed 2019-10-23 by MAKOMO

DennieSch, 2019-10-23:
Hi Marko, 
Drop-down looks good but... I noticed an anomaly or two. See attached. 
[Events Window (Buttons).zip](https://github.com/artisan-roaster-scope/artisan/files/3764383/Events.Window.Buttons.zip)

I maybe wrong but don't recall seeing them in previous versions/commits.

Den

DennieSch, 2019-10-24:
**UPDATE** - The problem with the text not respecting row height doesn't was happening when I was running with two monitors. It does not happen when running without an external monitor. Maybe, it has to do with my configuration (graphics card  and monitor). I'm using a MS Surface (first version) and the monitor was using the HDMI port of a TV.

The other problem is still there. I did notice that in addition to resizing the window,  resizing any of the columns clears/corrects the issue too.

MAKOMO, 2019-10-24:
I observed that "other problem" which seems to be caused by the version of the underlying GUI lib we are currently using. Earlier versions did not show this issue (without any code changes on the Artisan side). We have to hope that this will be fixed in the next version of the GUI lib Artisan is using. On the Artisan side there is nothing that can be done to prevent this.