artisan-roaster-scope/artisan #412: TP event not included in roast events or roast data table

DennieSch, 2019-10-22:
## Expected Behavior
TP event included in roast events (Roast/Roast Properties - Events)
TP event included in roast data table (Roast/Roast Properties - Data)

## Actual Behavior
TP event not included in roast events or roast data table
(see attached)

## Steps to Reproduce the Problem

  1. Perform roast
  2. Examine Roast/Roast Properties - Events
  3. Examine Roast/Roast Properties - Data

## Specifications

[turning point.zip](https://github.com/artisan-roaster-scope/artisan/files/3753815/turning.point.zip)

version = 2.0.1 (0b6f5c23)
Architecture = ('64bit', 'WindowsPE')
Machine = AMD64
Platform name = Windows-10-10.0.18362-SP0
Processor = Intel64 Family 6 Model 58 Stepping 9, GenuineIntel
Python Branch = tags/v3.7.4
Python Build = ('tags/v3.7.4:e09359112e', 'Jul 8 2019 20:34:20')
Python Compiler = MSC v.1916 64 bit (AMD64)
Python Implementation = CPython
Python Revision = e09359112e
Python version = 3.7.4
Release = 10
System = Windows
Version = 10.0.18362
Win32 = ('10', '10.0.18362', 'SP0', '')
  - Connected devices or roasting machine: none
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2019-10-23:
The TP event is kind of special. Everything recorded by the BT probe up to about DRY is an illusion. The probe is lagging behind the real bean temperature a lot in this part of the roast and the initial dip and rise of BT (including the TP) is fake. The beans start at around room temperature and their temperature only rises. There are no plans to add TP to any other places and we rather remove the ability to mark it on the graph in a future version.

closed 2019-10-23 by MAKOMO

DennieSch, 2019-10-23:
Yeah, I see what you mean. I hadn't thought of that way. TP is really the point at which a dropping probe temperature 'meets' a rising bean temp (and go on their merry way together from there :-). 
It is kind'a a fun point (albeit fake)  to reference as the 'beginning' of things.

MAKOMO, 2019-10-24:
Not exactly. The real bean temperature meets the temperature reported by the bean probe a little later (around 150C on most systems). TP is very bad as reference point for anything as it, well, fake.