artisan-roaster-scope/artisan #413: Roast Data - Copy Table; Display of negative time values in Excel

DennieSch, 2019-10-22:
## Expected Behavior
Properly display the values recorded

## Actual Behavior
Time values earlier than -1:00 are displayed as "0" or "#VALUE!".
Also, some values (XXX) are not displayed as decimals (XXX.0) as are the majority of the other values.
(see attached)

## Steps to Reproduce the Problem

  1. Open past roast
  2. Click 'Copy Table' on Roast/Roast Properties - Data
  3. Open Excel and Paste (Paste produces Circular Reference Warning)

## Specifications

[Copy Table - Data.zip](https://github.com/artisan-roaster-scope/artisan/files/3753944/Copy.Table.-.Data.zip)

version = 2.0.1 (0b6f5c23)
Architecture = ('64bit', 'WindowsPE')
Machine = AMD64
Platform name = Windows-10-10.0.18362-SP0
Processor = Intel64 Family 6 Model 58 Stepping 9, GenuineIntel
Python Branch = tags/v3.7.4
Python Build = ('tags/v3.7.4:e09359112e', 'Jul 8 2019 20:34:20')
Python Compiler = MSC v.1916 64 bit (AMD64)
Python Implementation = CPython
Python Revision = e09359112e
Python version = 3.7.4
Release = 10
System = Windows
Version = 10.0.18362
Win32 = ('10', '10.0.18362', 'SP0', '')

  - Connected devices or roasting machine: none
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

roasterdave, 2019-10-22:
The raw data copied to the clipboard is fine.  Both issues are Excel related.  

Negative time values are not handled as you expect in Excel.  For example, try entering "-3:00", "-3:06" and "-3:15" (without the quotes) into Excel.  You will get three different results.

Use the following method to paste data copied from Artisan into Excel.  Under the Paste icon on the Home menu in Excel choose "Use text import wizard".  Select "Delimited" and press Next.  Select "Tab" and press Next.  In the Data Preview area click on the Time column to select it then choose "Text" for the Column data format and press Finish.  

Values that don't show a decimal are just whole numbers.  Excel's default "General" format displays them this way.  You can change that by using the number format features in Excel.  You can also change the column formats in step three of the text import wizard.

closed 2019-10-22 by roasterdave

DennieSch, 2019-10-22:
Hello Dave- Thanks for the quick reply!

I see what you mean regarding Excel's handling of negative time values. What I recall from Excel is, to enter values/data as text, just proceed the value/data with an apostrophe ('-3:00, '3:06 and '3:15). So... I tried that with the examples you mentioned and they all did display correctly. I don't know if it is worth considering adding the apostrophe when the data is copied to the clipboard but... thought I would mention it. That way the data would look 'good' if someone just does the straight 'Copy Table' --> 'Paste' action. Just a thought.

I'm using a fairly old version of Excel so that may be the reason the Import Wizard isn't available as an option under Paste for me (the import wizard it is available under my 'Get External Data'). So... I just pasted the copied table into Notepad and imported it from there. Following your instructions, the time did import correctly! So that's a way to do it too.

Understand the decimals and formatting the precision after importing. Not a big deal, just thought it might look better (cosmetically) if all the data were formatted to the same precision upon paste/import. 

All the best!

Cheers,
Dennie

roasterdave, 2019-10-22:
Thank you too for the very fast test and feedback on this new feature!  

I am hesitant to add the leading apostrophe because other users may paste into something other than Excel and then have to edit them out. 

Thanks again for the report.

DennieSch, 2019-10-23:
Hey Dave- No problem, you're very welcome.

Understand your hesitation on the apostrophe and support your decision. 

I also tried this (and Tabular Text) in Events and Devices as well and they all worked fine for me.

Cheers,
Dennie