artisan-roaster-scope/artisan #414: Config|Colors|Graph Tab

DennieSch, 2019-10-29:
## Expected Behavior

Presence of a Cancel button
Color and code depicted for Title, Y Label and X Label

## Actual Behavior

No Cancel button present

Color and code not depicted for Title, Y Label and X Label. This appears to only occur when selecting/loading the Classic theme (Config|Themes|Artisan|Classic).

## Steps to Reproduce the Problem

  1. Select Classic theme
  2. Select Config|Colors
  3. see attached

[colors graph tab.zip](https://github.com/artisan-roaster-scope/artisan/files/3781878/colors.graph.tab.zip)

## Specifications

version = 2.0.1 (0a9f5e1e)
Architecture = ('64bit', 'WindowsPE')
Machine = AMD64
Platform name = Windows-10-10.0.18362-SP0
Processor = Intel64 Family 6 Model 58 Stepping 9, GenuineIntel
Python Branch = tags/v3.7.4
Python Build = ('tags/v3.7.4:e09359112e', 'Jul 8 2019 20:34:20')
Python Compiler = MSC v.1916 64 bit (AMD64)
Python Implementation = CPython
Python Revision = e09359112e
Python version = 3.7.4
Release = 10
System = Windows
Version = 10.0.18362
Win32 = ('10', '10.0.18362', 'SP0', '')
  - Connected devices or roasting machine: none
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2020-02-26:
The missing color name labels got fixed. We are aware that this dialog does not has a Cancel button/action as some other dialogs. We do not feel currently that it is worse the work to add one.

closed 2020-02-26 by MAKOMO

DennieSch, 2020-02-26:
Looks good. I did notice that if one has saved a theme where this issue was present, one should 're-save' the theme after re-applying the Classic theme then... all is good.
Understand on the Cancel button; no worries.