artisan-roaster-scope/artisan #415: File|Open Recent -- Enhancement

DennieSch, 2019-10-29:
## Expected Behavior

Ability to clear the Open Recent List
Ability to set a desired length for the Open Recent List

## Actual Behavior

Capability not currently available

## Steps to Reproduce the Problem

Thoughts for inclusion – 
- add ‘Clear Recent List’ 
- add ‘Show this number of Recent Documents:  (box with up/down spinners) ‘ 
- maybe... add a new item (Recent Documents...) to Config to handle these features.

## Specifications

version = 2.0.1 (0a9f5e1e)
Architecture = ('64bit', 'WindowsPE')
Machine = AMD64
Platform name = Windows-10-10.0.18362-SP0
Processor = Intel64 Family 6 Model 58 Stepping 9, GenuineIntel
Python Branch = tags/v3.7.4
Python Build = ('tags/v3.7.4:e09359112e', 'Jul 8 2019 20:34:20')
Python Compiler = MSC v.1916 64 bit (AMD64)
Python Implementation = CPython
Python Revision = e09359112e
Python version = 3.7.4
Release = 10
System = Windows
Version = 10.0.18362
Win32 = ('10', '10.0.18362', 'SP0', '')

  - Connected devices or roasting machine: none
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2020-05-19:
very low on our list

DennieSch, 2020-05-19:
Understand; I've been cleaning up the list, when needed, via regedit