artisan-roaster-scope/artisan #419: File|Report|Roast (PDF Option?)

DennieSch, 2019-10-29:
## Expected Behavior
Generate a 'Roast Report' directly to PDF

## Actual Behavior
Roast Report is generated in web/html format only, no option for PDF is currently provided.
I think it would be helpful/useful (at least for me :-) ) to be able to generate a roast report directly to PDF. (Currently I'm generating the report (in html) then, in the browser doing a print, change from default printer to PDF and generate the PDF. 
Maybe Roast could have a feature similar to Batches and Ranking -- like; 
Roast --> 
---- PDF...
---- Web...

## Steps to Reproduce the Problem

  1. Select File|Report|Roast
  2. 
  3. 

## Specifications

version = 2.0.1 (0a9f5e1e)
Architecture = ('64bit', 'WindowsPE')
Machine = AMD64
Platform name = Windows-10-10.0.18362-SP0
Processor = Intel64 Family 6 Model 58 Stepping 9, GenuineIntel
Python Branch = tags/v3.7.4
Python Build = ('tags/v3.7.4:e09359112e', 'Jul 8 2019 20:34:20')
Python Compiler = MSC v.1916 64 bit (AMD64)
Python Implementation = CPython
Python Revision = e09359112e
Python version = 3.7.4
Release = 10
System = Windows
Version = 10.0.18362
Win32 = ('10', '10.0.18362', 'SP0', '')

  - Connected devices or roasting machine: none
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2019-11-06:
This is not easily possible as Artisan just generate HTML here for simplicity to let the browser do the rendering and eventual the export to PDF or any other format.

closed 2019-11-06 by MAKOMO

InvisibleOrdinary, 2019-11-06:
You can (should be able to) print to pdf out of your browser, if that helps.

DennieSch, 2019-11-06:
Thanks! Actually, that is what I have been doing. Just thought it would be useful to be able to generate the 'Report' directly to PDF. One can to a 'Convert To' and 'Save Graph' directly to PDF but that doesn't include all the other attributes of the roast. I think the Report is very useful since it captures all the key attributes of the roast (and cupping) in one file which can be viewed, printed, filed, archived, etc. outside of Artisan.