artisan-roaster-scope/artisan #420: Roast Properties|Ambient Source

DennieSch, 2019-11-04:
## Expected Behavior
Clicking 'Update' loads temperature into 'Ambient Conditions' 

## Actual Behavior
Temperature does get loaded but only if "Phidget TMP1101 4xTC AT" is not added under Device Assignment|Extra Devices

See attached

[Ambient Source.zip](https://github.com/artisan-roaster-scope/artisan/files/3802664/Ambient.Source.zip)

## Steps to Reproduce the Problem

  1. Add Phidget TMP1101 4xTC AT as and 'Extra Device'
  2. Click 'Update' on Roast Properties 
  3. Observe no temperature is loaded into Ambient Conditions

## Specifications

version = 2.0.1 (6b5f6736)
Architecture = ('64bit', 'WindowsPE')
Machine = AMD64
Platform name = Windows-10-10.0.18362-SP0
Processor = Intel64 Family 6 Model 58 Stepping 9, GenuineIntel
Python Branch = tags/v3.7.4
Python Build = ('tags/v3.7.4:e09359112e', 'Jul 8 2019 20:34:20')
Python Compiler = MSC v.1916 64 bit (AMD64)
Python Implementation = CPython
Python Revision = e09359112e
Python version = 3.7.4
Release = 10
System = Windows
Version = 10.0.18362
Win32 = ('10', '10.0.18362', 'SP0', '')

- Connected devices or roasting machine: Phidgets HUB0000_0   and  TMP1101_0
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2019-11-04:
Correct observation. The whole ambient temperature collection thing is not very well thought out, was added years ago and modified over and over as more hardware popped up. Originally, the idea was that one could choose one extra device channel from the popup from which the ambient temperature values is computed by averaging all available values. Later the harvesting of the internal temperature sensor of Phidget TC modules was added and even later the direct support of the Ambient Phidgets HUM1000/PRE1000 as well as the corresponding Yoctopuce modules.

I updated the behavior now such that
- the update function is again automatically triggered at DROP
- the update function collects data from the selected temperature curve if one is selected and otherwise tries to take the reading of a connected Phidget TC AT channel
- the collection of data from a Phidget TC AT channel should now work if this one is used as extra temperature channel or not

Note that data from Phidget Ambient modules is collected already on ON and that this mechanism works independent of the above ambient update function.

Hope that I did not mess up anything.

closed 2019-11-04 by MAKOMO

DennieSch, 2019-11-04:
Hi Marko,
Just wanted to provide you some feedback (all good, I think) of what I'm now seeing.

Using your descriptions (#) --

# I updated the behavior now such that

# the update function is again automatically triggered at DROP
---Yup

# the update function collects data from the selected temperature curve if one is selected and otherwise tries to take the reading of a connected Phidget TC AT channel
---Yup 
---If ON=TRUE and START=FALSE and selected temperature is NULL,  'Update' click/function is "...the reading of a connected Phidget TC AT channel". All other 'selected temperature' sources return "0"
---If ON=TRUE and START=TRUE then, "the update function collects data from the selected temperature curve if one is selected and otherwise tries to take the reading of a connected Phidget TC AT channel"

# the collection of data from a Phidget TC AT channel should now work if this one is used as extra temperature channel or not
---Yup (I removed the Phidget TC AT channel from Extra Devices and data is still collected)

# Note that data from Phidget Ambient modules is collected already on ON and that this mechanism works independent of the above ambient update function
---Yup

Again, just wanted to share with you the functionality I was seeing and what you intended... which is great!

Cheers,
Dennie

DennieSch, 2019-11-04:
Oops... didn't realize the annotation I used would result in the font in my previous post. Sorry, that was not my intent.  - Dennie