artisan-roaster-scope/artisan #421: ranking/web stretches background logo

InvisibleOrdinary, 2019-11-06:
## Expected Behavior

logo is shown as logo in correct proportions

## Actual Behavior

background logo is stretched

## Steps to Reproduce the Problem

  1. open .alog file(s) via file/ranking/web…
  2. 
  3. 

## Specifications

  - Artisan Version: 2.0.1 (95be8ff) 
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version): OSX
  - Connected devices or roasting machine: none for this
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

[Proster-10K.aset.zip](https://github.com/artisan-roaster-scope/artisan/files/3812237/Proster-10K.aset.zip)

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

InvisibleOrdinary, 2019-11-06:
actually, it also does it when the graph area resizes. if the sliders are hidden, and then turned on (like when you hit 'on'. or anything else that re-scales the graph. It stretches the logo in the background.

roasterdave, 2019-11-10:
Thanks for reporting this case.  Closing as this issue is fixed in the commit referenced above.

closed 2019-11-10 by roasterdave