artisan-roaster-scope/artisan #423: Phidgets devicesconly connecting when unplugged/re-plugged in

InvisibleOrdinary, 2019-11-06:
## Expected Behavior

(all) Phidgets readings available when on/started

## Actual Behavior

Phidgets seem to only be available the first time they are plugged in, and not after. If I select 'on' when the phidgets are connected already, i get -1 readings. inconsistently for 1 or all. I see no pattern to which. 

they appear when I unplug the usb cable, and re-insert it. then it goes through the process of connecting and all device readings show. 

If I stop 'end' the roast and re-start (or 'on' it doesn't seem to matter which), then they are back to -1 readings.

This has been working fine with two temp probes and all others listed below:

I tested this several time to make sure it wasn't a loos wire or something else. And it always needed re-inserted to be recognized

Things that have changed: 
 - I added 1 temp probe (so when from 2 to 3 out of 4 available). 
 - made some minor adjustments to the symbolic devices.
 - made some questionable color choices for the curves

_______
ADDITIONAL NOTES
I checked with the phidgets control panel, and all devices read as expected with no issue. 
I checked the connections - both visibly checking the physical connections and by switching probes to different ports  - All worked as expected in the phidgets control panel, and did not in artisan
I did not test on a more recent beta release. The last one (continuous build) that I have seen available that included a mac version didn't work at all, so went back to this version
I did not test on the most recent stable build as this was working fine up to this point.

## Steps to Reproduce the Problem

 see above

## Specifications

  - Artisan Version: 2.0.1 (95be8ff) 
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version): MAC OS 10.14.6
  - Connected devices or roasting machine: Phidgets:
VINT Hub with: 
- TMP1101_0 with quick connects in all 4 ports. 0-3 actually connected to K-type temp probes
 - PRE1000_0
 - HUM1000_0
 - 1137_0 (differential pressure ±7 kPa)

Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.

[Deft-Pinhalense-Sampler.aset.zip](https://github.com/artisan-roaster-scope/artisan/files/3815582/Deft-Pinhalense-Sampler.aset.zip)

Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2019-11-07:
Dear Dave,

could you please test the current beta builds if those work any better for you and report back?

https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous

Thanks,
Marko

MAKOMO, 2019-11-09:
Is this resolved or not?

InvisibleOrdinary, 2019-11-09:
No. But maybe. I haven’t seen a new build that included a mac version since 94daacb

MAKOMO, 2019-11-09:
Hm. Sometimes the Mac builds are missing. Must be a timing issue in the CI process. Currently there is a Mac build available that should fix this one and the ranking report dark-mode issue you reported.

MAKOMO, 2019-11-12:
Just found more troubles with Phidget connects and disconnects. I just pushed a new commit that should resolve them. Could you please test on your setup and report?

MAKOMO, 2019-11-12:
Builds may need still 20min until they are online

MAKOMO, 2019-11-24:
Still unconfirmed if the recent builds resolve this. Could you please comment. This one is delaying the v2.1 release.

MAKOMO, 2019-11-25:
No further input on this and this is reported by some as resolved. I close it.

closed 2019-11-25 by MAKOMO