artisan-roaster-scope/artisan #425: background logo doesn't retain opacity

InvisibleOrdinary, 2019-11-09:
## Expected Behavior

logo in background keep opacity set prior to closing artisan

## Actual Behavior

it appears that it is setting the opacity at 100%, but keeping the number you have set in the UI tab. So that if you sey it at 20%, when reopening the logo is at 100%, but the setting is 20 still - and you have to set it t0 2% to get the same opacity…

## Specifications

  - Artisan Version: 2.0.1 (94daacb) 
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version): Mac 10.14.16
  - Connected devices or roasting machine: 
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

roasterdave, 2019-11-09:
I am not able to reproduce this problem.  Can you provide a step-by-step procedure to follow?

MAKOMO, 2019-11-25:
I cannot reproduce this either.

closed 2019-11-25 by MAKOMO