artisan-roaster-scope/artisan #427: File|Report|Ranking|Web... Stacked bar graph; low contrast of text color

DennieSch, 2019-11-12:
## Expected Behavior
Higher readability of text in stacked bar graph

## Actual Behavior
Low contrast of text color vs. bar color makes readability of the numbers problematic.

See attached 
[stacked bar graph - text contrast.zip](https://github.com/artisan-roaster-scope/artisan/files/3834309/stacked.bar.graph.-.text.contrast.zip)

## Steps to Reproduce the Problem

  1. Generate a Ranking Report (web)
  2. Review stacked bar graph at bottom of report
  3. 

## Specifications

version = 2.0.1 (c856429f)
Architecture = ('64bit', 'WindowsPE')
Machine = AMD64
Platform name = Windows-10-10.0.18362-SP0
Processor = Intel64 Family 6 Model 58 Stepping 9, GenuineIntel
Python Branch = tags/v3.7.5
Python Build = ('tags/v3.7.5:5c02a39a0b', 'Oct 15 2019 00:11:34')
Python Compiler = MSC v.1916 64 bit (AMD64)
Python Implementation = CPython
Python Revision = 5c02a39a0b
Python version = 3.7.5
Release = 10
System = Windows
Version = 10.0.18362
Win32 = ('10', '10.0.18362', 'SP0', '')

  - Connected devices or roasting machine: n/a
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2019-11-12:
Thanks for reporting! This should be improved a bit with the commit 41470488934e851fcdd0ba70e0d519913de0daee.

closed 2019-11-12 by MAKOMO

DennieSch, 2019-11-12:
Much better... looks great!