artisan-roaster-scope/artisan #428: Communication error Siemens S7 Kirsch und Mausser UG Retro

FORTHOM, 2019-11-22:
Dear all, we are new users of Artisan and are very happy with the performance of this great software. Unfortunately it doesn't work reliable on our machine/system.
Thank so much you for your support (however, we have no idea of programming computers ;-) )

## Expected Behavior
Connection of Roasting Machine and Computer via LAN; After starting, all relevant parameters are displayed and recording can be started without problems however:

## Actual Behavior
sporadically (first after about 30 min) Error "S7 Communication Error readFloat: Communication error". After resetting starting a new rocord fails: "S7 Communication Error readFloat: [Errno 24] Too many files open: 'nul'". After restarting Artisan it works well until the error occurs again...

## Steps to Reproduce the Problem

  1. Re-Booting Roaster/Laptop
  2. Using another Computer/LAN cable
  3. Ping works fine over several ours

## Specifications

  - Artisan Version: 2.0.1
  - Artisan Build (number in brackets shown in the about box):  (a10825bd)
  - Platform (Mac/Windows/Linux + OS version): WIN10
  - Connected devices or roasting machine: Siemens S7/UG 22 Retro
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2019-11-23:
This is a tricky one as it is almost impossible to reproduce for us. Here is my guess what happens: the communication with the UG experiences random communication errors and on some of those the connection is closed and restarted again (without being observed by you). However, those closed connections do not release their "file handle" and thus the number of open files increases until the Windows runs out of handles. Strange that nobody reported this before as there are a number of users that use the S7 communication with Artisan on different machines incl. the UG 22 Retro.

I reviewed the corresponding code in the S7 module of Artisan and applied two small changes that could prevent those closed connections to keep their file handles open. However, the problem might very well be also in one of the underlying S7 libraries that Artisan is using.

So I would like to ask you to test the new v2.0.1 builds (beta versions of the upcoming v2.1) available here:

https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous

and report if this works better for you.

Even if my change allows  you to roast longer than 30min, I would like to check on those communication errors too. For this, could you please export your Artisan settings via menu Config >> Help >> Save Settings as .aset file, change that file extension into .txt and attach it here via drag-and-drop. I would also like to see your error log after you have done one roast. Just copy what is in the error log window (menu Help >> Errors) into a comment here.

Thanks for your support!

FORTHOM, 2019-11-25:
Dear Marko, thanks a lot for your immediate support.
To reproduce the error with the old build, I again started a reading. After 19 min I got the error message as known (find attached the respective files).
I then installed the new build, however, after 13 min the system stopped with the same error (files attached).
I also tried to save the respective data-files for you but failed in both cases due to the same "too many files.."-error.

I'm looking forward to hearing from you, hopefully with a new idea what could have happend ;-)

Thank you!
[settings_old.txt](https://github.com/artisan-roaster-scope/artisan/files/3886071/settings_old.txt)
[errorlog_new.txt](https://github.com/artisan-roaster-scope/artisan/files/3886073/errorlog_new.txt)
[systemlog_new.txt](https://github.com/artisan-roaster-scope/artisan/files/3886074/systemlog_new.txt)
[settings_new.txt](https://github.com/artisan-roaster-scope/artisan/files/3886075/settings_new.txt)
[systemlog_old.txt](https://github.com/artisan-roaster-scope/artisan/files/3886076/systemlog_old.txt)
[errorlog_old.txt](https://github.com/artisan-roaster-scope/artisan/files/3886077/errorlog_old.txt)

MAKOMO, 2019-11-25:
Ich sehe Du sprichst Deutsch. Wo sitzt Du denn ungefähr? 
Ich sehe da jede Sekunde einen Kommunikationsfehler. Funktioniert die Aufzeichnung überhaupt? Könntest Du mir Bitte ein aufgezeichnetes .alog File schicken?
Du sagst ja, es funktioniert so die ersten Minuten ganz gut. Könntest Du es mal mit einem Abtastinterval von 2sec statt 1sec probieren (Menu Konfiguration >> Abtastinterval)?

FORTHOM, 2019-11-25:
Unsere Rösterei ist in Fürth ("Fortezza-Espresso"). Anbei eine alog-Datei. Tatsächlich gelingt es ja zwischendurch eine komplette Röstung - aus meiner Sicht auch ohne jede Einschränkung - aufzuzeichnen. Das Abtastintervall haben wir schon auf 5s hochgesetzt, allerdings ohne einen Effekt zu sehen.
[19-11-09_1216.txt](https://github.com/artisan-roaster-scope/artisan/files/3886293/19-11-09_1216.txt)

MAKOMO, 2019-11-25:
Ich hab nun den S7 Support etwas umgebaut. Es scheint ein Problem in der eigentlichen S7 Bibliothek zu geben die Artisan meldet, dass die Verbindung noch steht, obwohl das Gerät die Verbindung wegen eines Fehlers beendet hat. Ich hab da nun drum rum gearbeitet. 

Meine Vermutung ist nun, dass es ab und an Kommunikations Probleme gibt (so was ist eher normal wenn man neben starken Elektromotoren arbeitet und nicht beunruhigend) bei deren Auftreten die Maschine die Verbindung trennt und Artisan dies nicht mitbekommt. Wenn man dann Artisan aus und wieder einschaltet, versucht Artisan wiederholt eine neue Verbindung aufzubauen, dies geht aber schief da ja noch eine offene (aber kaputte) Verbindung besteht.

Könntest Du noch einmal probieren, ob meine weiteren Änderungen die Situation besser behandeln?
Hier gibt es neue Builds: https://github.com/artisan-roaster-scope/artisan/releases

Danke!

FORTHOM, 2019-11-26:
Hallo Marko, zunächst nochmal besten Dank für dein jeweils promptes Engagement unser Problem betreffend. Auch mit dem neuen Build besteht der Fehler aber weiterhin.
Ich habe mich aber nochmal mit dem Abtastintervall systematischer beschäftigt: 1s: Abbruch nach ca. 7min, 2s: Abbruch nach ca. 13 min, 10s: Lief nach 20min noch "normal". Interessant: Mit der 2s-Einstellung habe ich eine Aufzeichnung nach 7min manuell gestoppt und abgespeichert sowie nach "Reset" eine neue Aufzeichnung gestartet. Diese brach dann schon nach 7min ab. Es sieht also danach aus, dass sich die Fehler im Hintergrund summieren und erst nach Neustart von Artisan "auf Null" zurückgesetzt werden (deshalb war ich auch zunächst der Meinung, dass das Hochsetzen der Zeit keinen Effekt hat - da hatte ich aber die Software vorher nicht neu gestartet). Wenn wir mit 3 oder 4s arbeiten und Artisan nach jeder Aufzeichnung neu starten, könnte es also funktionieren. Da in der Routine unsere Batches aber releativ eng getaktet sind, wäre das nicht sehr komfortabel.... Ich kann das morgen nochmal in der Praxis testen.

MAKOMO, 2019-11-26:
Danke fürs erneute Testen. 

> 
> Hallo Marko, zunächst nochmal besten Dank für dein jeweils promptes Engagement unser Problem betreffend. Auch mit dem neuen Build besteht der Fehler aber weiterhin.

Blöd. Ich werde das was ich nun habe dennoch als v2.1 Heute veröffentlichen.

> Ich habe mich aber nochmal mit dem Abtastintervall systematischer beschäftigt: 1s: Abbruch nach ca. 7min, 2s: Abbruch nach ca. 13 min, 10s: Lief nach 20min noch "normal". Interessant: Mit der [...]

Ja, die Fehler summieren sich auf und werden, warum auch immer, nicht korrekt behandelt. Nach jeder Röstung die Software Neustarten ist keine wirklich schöne Lösung. Wir müssen der Sache auf den Grund gehen und direkt lösen. Das braucht wohl etwas mehr Zeit. Zunächst muss ich verstehen, was da genau passier. Ich werde Dir in den nächsten Tagen eine Version basteln, welche genauer protokolliert was passiert.

Komisch, dass sonst niemand über die Jahre von solch einem Problem berichtet hat. Artisan ist mit mehreren Kirsch Maschinen problemlos im Betrieb. Eventuell treten da weniger Kommunikations-Störungen auf. Was mich auch verwundert hat, ist das bei Dir einige Werte immer auf Null sind (). Ich glaube es waren GasP und DrumP und noch ein Wert. 

M

FORTHOM, 2019-11-26:
Noch eine - vielleicht etwas unfachmännische - Frage/Anmerkung: Könnte man die Software über den Reset-Button, der ja zwischen zwei Röstungen ohnehin betätigt werden muss, soweit zurücksetzen, dass es einem Neustart quasi analog ist? Oder spielt da Windows bzw. andere Funktionen von Artisan nicht mit? Ansonsten wäre das ein zunächst praktikabler Weg, wenngleich damit natürlich die Ursache des Problems nicht behoben wird.
Dass einige Werte auf Null sind wundert mich eigentlich nicht, da wir beispielsweise die DrumSpeed nicht regeln können. Da sind die Maschinen werksseitig individuell konfiguriert.
Übrigens noch etwas: Der Röster selbst lief bei meinen heutigen Tests gar nicht; nur die Steuerung (die Temp.sensoren haben insoweit halt Raumtemperatur angezeigt)...

MAKOMO, 2019-11-26:
Ja, dies wäre die Idee, die Verbindung komplett zurückzusetzen (auch im Fehlerfalle). Leider scheint dies nicht zu klappen, da irgendetwas in der drunterliegenden Lib hängt. Ich kann das auch auf meinem  Mac nicht nachvollziehen. Eventuell passiert das nur auf Windows. Bin Heute und Morgen recht eingespannt, schau mir das aber dann Ende der Woche noch genauer an. Wir sollten das offline nehmen. Kannst Du mir über das Contact Formula hier <https://artisan-roasterscope.blogspot.com/p/contact-me.html> ne Nachricht schrieben, dann antworte ich per EMail. Danke!

MAKOMO, 2019-11-28:
Ich hab das Problem vermutlich gefunden und hoffentlich gelöst. Es liegt gar nicht an der S7 Kommunikation, sondern an der Programmumgebung welche unter Artisan liegt. Ein sehr grundlegendes Problem das nur auf Windows in dieser neuen Artisan Version auftritt. Könntest Du auf die Schnelle mal die Version 2.1.1 hier testen?  https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous

Danke!

FORTHOM, 2019-11-29:
Im "Trockentest" (nur mit laufender Steuerung, ohne aktiven Röster) sieht es sehr gut aus! Trotz Abtastintervall von 1s kein Abbruch bis jetzt. Nächste Woche werden wir nochmal im Realbetrieb testen und berichten.

Vielen Dank bis hierher. Top Unterstützung!!

PS: Hatte über Artisan-"contact" noch eine Frage zur direkten Steuerung des Rösters (ich hatte gehofft, dass wir alle "Aktionen", also z.B. Brennersteuerung, über Artisan machen können. Ggf. kannst du da nochmal - per mail - darauf eingehen?

MAKOMO, 2019-11-29:
Danke für das Testen! Mir ist gerade eben bestätigt worden, dass das Problem tatsächlich beseitigt wurde. Es sollte also auch im Realbetrieb funktionieren. Ich werde später Heute genau dieses Build als v2.1.1 veröffentlichen.

Leider hab ich Deine Anfrage per Contact Formular nicht bekommen. Ist auch nicht in meinem Spam aufgetaucht. Ich werde Euch über die Email von Eurer Seite eine Email schreiben und bitten noch einmal die Frage wegen der Brennersteuerung zu stellen.

Grüße
Marko

MAKOMO, 2019-11-29:
Email an info@ geschrieben.

closed 2019-11-29 by MAKOMO

MAKOMO, 2019-11-29:
Das Artisan Kontaktformular funktioniert momentan nicht. Google kennt das Problem wohl schon seit einem Monat hat aber noch keine Lösung dazu angeboten:(

Bezüglich der UG Steuerung. Artisan kann potenziell jede per MODBUS oder S7 angeschlossene Maschine komplett steuern. Allerdings habe ich keine Informationen welche Register die UG dafür vorsieht. Eventuell erlaubt die PLC dies auch gar nicht. Am Besten Du fragst da mal beim Hersteller nach. Ich wäre bereit die entsprechende Konfiguration für Artisan zu entwerfen, falls ich die nötigen Informationen (also insbesondere die Register) dazu bekomme und jemand bereit ist das zu testen.