artisan-roaster-scope/artisan #429: Modbus Communication Error, Help me please

SLEEPENEMY, 2019-11-22:
## Expected Behavior

## Actual Behavior
Install Artisan > load setting {artisan-settings - salon 2.3 touch screen} 
> Set up IP Address 192.168.1.99 > Insert eternal cable > select machine; Coffee Tech silon ZR7 
> set IP address 192.168.1.27 > select ON 

## Steps to Reproduce the Problem

  1. reinstall and run the same steps
  2. on / off roasting machine + IMAC + check cable
  3. still appear error
[Test Artisan.zip](https://github.com/artisan-roaster-scope/artisan/files/3879028/Test.Artisan.zip)
[artisan-settings - silon 2.3 touch screen.zip](https://github.com/artisan-roaster-scope/artisan/files/3879029/artisan-settings.-.silon.2.3.touch.screen.zip)
[artisan-settings - silon 2.3 touch screen.aset.txt](https://github.com/artisan-roaster-scope/artisan/files/3879031/artisan-settings.-.silon.2.3.touch.screen.aset.txt)

## Specifications

  - Artisan Version: 2.0
  - Artisan Build (number in brackets shown in the about box): 5.12.3.
  - Platform (Mac/Windows/Linux + OS version): IMAC late 2012 OS High Sierra 10.13.6
  - Connected devices or roasting machine: Coffee Tech Silon ZR7
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2019-11-22:
Did you contact CTE on this? I have not much information on this. It seems Artisan is not able to communicate with your machine at all.

SLEEPENEMY, 2019-11-23:
Hi,
I will contact CTE.

Thank you very much

MAKOMO, 2019-11-24:
Seems not to be an implementation issue

closed 2019-11-24 by MAKOMO