artisan-roaster-scope/artisan #430: Auto file save protocol

stinsonddog, 2019-11-22:
## Expected Behavior

Autosave - current Prefix>Batch #> Date

## Actual Behavior
Autosave -  current Prefix>Batch #> Date
## Steps to Reproduce the Problem

N/A

## Specifications

  - Artisan Version:  for 2.2
  Modify current Autosave profile behavior.

Allow user to specify through numbering the order and content of the file save attributes:

Batch Counter Number
Prefix
Date
Coffee Name
User option text
User option text

This would enable users to avoid Save As, and speed up the saving process.  The numbering allows flexibility in file name conventions.  

Open to other ideas on how to enhance them.  

See some discussion here:
https://www.home-barista.com/home-roasting/saving-artisan-profiles-naming-saving-file-location-etc-t61713.html#p684756

DennieSch, 2019-11-28:
I too would like lend my support to this request ... to see some additional options or flexibility in how file names for profiles are composed or generated.
I provided some discussion on what I do with file names here -- 
https://www.home-barista.com/home-roasting/saving-artisan-profiles-naming-saving-file-location-etc-t61713-10.html#p685157

InvisibleOrdinary, 2019-12-05:
I third this.
Having the ability to name files with data so that it's searchable in file systems would be great. Currently you can do that, but after the fact, and artisan doesn't recognize those changes, so it if you use a roast as a profile, or reopen it to review, you get the original name and not the renamed name. 
I'm on mac, but I assume this behavior is the same on other OSs.

MAKOMO, 2020-04-04:
added in Commit 98fe39b9f7eb7f7c73dd8132c99583258df2c9b9

closed 2020-04-04 by MAKOMO