artisan-roaster-scope/artisan #433: 2.0.1 Event Slider/Buttons VOUT Command: out(2,{}) not working

arkenamun, 2019-11-25:
## Expected Behavior

Roaster electrical heating element controlled by VOUT Command: out(2,{})

## Actual Behavior

No output, no errors. Reverted to 2.0.0, now functions correctly.

## Steps to Reproduce the Problem

Start roast, turn power to 100%

## Specifications

 Artisan Version: 2.0.1
Artisan Build (number in brackets shown in the about box): sorry, reverted to 2.0.0 already, downloaded 21/11/2019
Platform (Mac/Windows/Linux + OS version): W10
Connected devices or roasting machine: Custom config, Proaster sample roaster

[v2.0.1 aborted warm-up. no power. 19-11-25_1310.alog.txt](https://github.com/artisan-roaster-scope/artisan/files/3884979/v2.0.1.aborted.warm-up.no.power.19-11-25_1310.alog.txt)
[Proaster v3.1.0 [autoCHARGE control].aset.txt](https://github.com/artisan-roaster-scope/artisan/files/3884980/Proaster.v3.1.0.autoCHARGE.control.aset.txt)

MAKOMO, 2019-11-25:
Which Phidget do you control via this VOUT command?

MAKOMO, 2019-11-25:
Did you see an attach message for your device in the message log (menu Help >> Messages) on activating that slider?

arkenamun, 2019-11-25:
1: Phidget Details

OUT1001 - Isolated 12-bit Voltage Output Phidget on HUB0000_0 Port 2

2: Attach Message

**v2.0.1 (814ae3bb)**

6 16:00:52.226 Phidget IO 6/6/6 attached

5 16:00:52.226 Phidget IO 6/6/6 attached

4 16:00:52.225 Phidget Temperature Sensor 4-input attached

3 16:00:51.761 Scope monitoring...

2 16:00:50.983 Scope has been reset

1 15:59:55.717 Background C:/Users/Roaster/Dropbox/Stash/Roasting/Artisan Data/Proaster/Proaster H - Profile/NI Washed P407_19-09-02_1731.alog loaded successfully 17:31

**v2.0.0:**

7 15:57:04.388 Phidget IO 6/6/6 attached

6 15:57:04.388 Phidget IO 6/6/6 attached

5 15:57:04.387 Phidget Temperature Sensor 4-input attached

4 15:57:03.898 Scope monitoring...

3 15:57:03.252 Scope has been reset

2 15:57:00.560 

1 15:56:59.997 Background C:/Users/Roaster/Dropbox/Stash/Roasting/Artisan Data/Proaster/Proaster H - Profile/NI Washed P407_19-09-02_1731.alog loaded successfully 17:31

**Note there are 2x HUB0000, removing the 2nd, still no output. Messages relating to detach and further attempt with single HUB0000 below..**

16 16:02:29.114 Event # 1 recorded at BT = 42.5 Time = 00:06

15 16:02:25.175 follow on

14 16:02:22.441 Scope recording...

13 16:02:15.324 Phidget Temperature Sensor 4-input attached

12 16:02:14.844 Scope monitoring...

11 16:02:14.163 Scope has been reset

10 16:02:00.861 Phidget IO detached

9 16:02:00.861 Phidget IO detached

8 16:02:00.758 Phidget Temperature Sensor 4-input detached

7 16:02:00.385 Scope stopped

6 16:00:52.226 Phidget IO 6/6/6 attached

5 16:00:52.226 Phidget IO 6/6/6 attached

4 16:00:52.225 Phidget Temperature Sensor 4-input attached

3 16:00:51.761 Scope monitoring...

2 16:00:50.983 Scope has been reset

1 15:59:55.717 Background C:/Users/Roaster/Dropbox/Stash/Roasting/Artisan Data/Proaster/Proaster H - Profile/NI Washed P407_19-09-02_1731.alog loaded successfully 17:31

arkenamun, 2019-11-25:
No events recorded in 2.0.1 message log records Event when using slider/buttons calling the VOUT Command.

Messages from 2.0.0:

16 16:02:29.114 Event # 1 recorded at BT = 42.5 Time = 00:06

15 16:02:25.175 follow on

14 16:02:22.441 Scope recording...

13 16:02:15.324 Phidget Temperature Sensor 4-input attached

12 16:02:14.844 Scope monitoring...

11 16:02:14.163 Scope has been reset

10 16:02:00.861 Phidget IO detached

9 16:02:00.861 Phidget IO detached

8 16:02:00.758 Phidget Temperature Sensor 4-input detached

7 16:02:00.385 Scope stopped

6 16:00:52.226 Phidget IO 6/6/6 attached

5 16:00:52.226 Phidget IO 6/6/6 attached

4 16:00:52.225 Phidget Temperature Sensor 4-input attached

3 16:00:51.761 Scope monitoring...

2 16:00:50.983 Scope has been reset

1 15:59:55.717 Background C:/Users/Roaster/Dropbox/Stash/Roasting/Artisan Data/Proaster/Proaster H - Profile/NI Washed P407_19-09-02_1731.alog loaded successfully 17:31

MAKOMO, 2019-11-25:
Thanks for this detailed report. This one is difficult for me to debug as I lack any VOUT Phidget hardware to test.

Could you please try to replace the out(2,{}) by out(0,{}) in the slider command and see if that works. The OUT1001 is a one channel VINT device and the channel number of the out command should match the channel of the actual module (which is 0 as it has only one channel). The HUB port that module is connected to should not play a role. I am investigating why your setup worked on Artisan v2.0. I am investigating....

MAKOMO, 2019-11-25:
What is your complete Phidget setup? You say you have two VINT HUBs!? One is connected to that OUT1001 to control the gas. I see also a TMP1101 and a HUM1000 and a PRE1000. Anything else?

arkenamun, 2019-11-25:
Yes, I have now simplified. I had both my sample roasters connected, which was unnecessary. Current connections are as below plus HUB5000_0 Wi-Fi for ambient with the HUM1000 and PRE1000

[image]

MAKOMO, 2019-11-25:
What happens if you use out(0,{}) instead of the out(2,{}) as the slider action?

arkenamun, 2019-11-25:
I shall check, I have to remove the chaff collector and control panel cover on the red sample roaster in this picture to get to the controls , it might take a while, but I shall do it today.

[image]

MAKOMO, 2019-11-25:
NO! Do not change the hardware setup. Just swap the slider command in software! So replace that out(2,{}) by out(0,{}) as that OUT1101 has only one port and you want to access the first one (number 0).

[image]

arkenamun, 2019-11-25:
Thank you for the clarification. I understand.

arkenamun, 2019-11-25:
Yes, that works thank-you!  The heating element is now turning on with slider and buttons. There are still no event messages.

closed 2019-11-25 by MAKOMO