artisan-roaster-scope/artisan #434: Errors opening an .alog (2.0 & 2.1)

DennieSch, 2019-11-27:
## Expected Behavior
Proper/nominal display of curves with events annotated.

## Actual Behavior
Event annotations missing (except for a user event)

After (re)loading settings, annotations now display but x-axis MIN is changed (maybe other things, not sure)

Errors are generated
Occurs with both 2.0 and 2.1
Seems to occur with only one of my .alog files. Not sure if this .alog is just corrupted or if it could be something else. Since the software did generate errors, thought I would pass this along in case it wasn't a corrupted .alog file. 

See attached

[Error Messages when loading an alog file.zip](https://github.com/artisan-roaster-scope/artisan/files/3895455/Error.Messages.when.loading.an.alog.file.zip)

## Steps to Reproduce the Problem

  1. Open .alog file
  2. Load settings
  3. 

## Specifications

version = 2.1.0 (73534dc3)  AND  2.0.0 (a0a1fac)
Architecture = ('64bit', 'WindowsPE')
Machine = AMD64
Platform name = Windows-10-10.0.18362-SP0
Processor = Intel64 Family 6 Model 58 Stepping 9, GenuineIntel
Python Branch = tags/v3.7.5
Python Build = ('tags/v3.7.5:5c02a39a0b', 'Oct 15 2019 00:11:34')
Python Compiler = MSC v.1916 64 bit (AMD64)
Python Implementation = CPython
Python Revision = 5c02a39a0b
Python version = 3.7.5
Release = 10
System = Windows
Version = 10.0.18362
Win32 = ('10', '10.0.18362', 'SP0', '')

  - Connected devices or roasting machine: N/A
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

roasterdave, 2019-11-27:
Thanks for reporting!  There is incomplete information about the extra devices in the .alog file.  How that happened may remain a mystery since the file was created with a pre-release build.  Pre-release builds often have issues.  

To permanently workaround the problem simply open the .alog file in v2.0 or v2.1 and then save it again (either to the same or a new file name).   The saved file will now open without error.

roasterdave, 2019-11-27:
Well, it seems my suggestion only works under certain circumstances.  Let me do some more investigating.

DennieSch, 2019-11-27:
Tried your suggestion with the following results. I believed it worked (recovered) THANKS!.

Just wanted to share the following in case it helps you in 'investigating' and/or if this is what you would expect.

Performed with v2.1.0 (73534dc3)
Open original alog file -- displayed with information missing (as before)
Save As (with a different file name 'A') 
Open the saved alog file 'A'  -- no change/difference; displayed with information missing
Load Settings  -- display is cleared; no information is displayed at all (this is different but appears to be normal...I think)

Open original alog file -- displayed with missing information (as before)
Load Settings  --  missing information now displayed along with x-axis MIN changed (as noted in my original posting)
Save As (with a different file name 'B' )
Reset (the Controls Reset Button)
Open the saved alog file 'B'  --  displayed information now looks normal (recovered)

Hope this is of some help (assuming this is still worth chasing down) otherwise, thank you for your help!

MAKOMO, 2019-11-28:
Excellent report! You were hit by a corner case where Artisan failed to correctly load a profile generated with different  extra device setup. I fixed this and the new 2.1.1 build available in half an hour should allow you to load this profile correctly.

closed 2019-11-28 by MAKOMO