artisan-roaster-scope/artisan #437: SC Start and SC End button functionality after Drop

DennieSch, 2019-12-05:
## Expected Behavior
SC Start and SC End buttons remain active (not grayed out) and functional until Cool End is selected/clicked.

## Actual Behavior
When Drop is selected/clicked, SC Start and SC End buttons are grayed out and no long available  to record their associated event. 

NOTE - In v2.0, SC Start and SC End buttons were grayed out after Drop but were still functional (event was recorded).

Roasting with a Behmor, my 'Drop' is switching to Cool on the roaster (not a physical drop). Sometimes roasting just into SC then, click Drop (and start Cool) or, SC may start shortly after Drop. SC then coasts into the cooling period for a bit and when SC stops I record this event (which was possible in v2.0 although the button was grayed out.)

## Steps to Reproduce the Problem

  1. Perform a roast
  2. Click'Drop'
  3. Notice SC Start and SC End buttons become grayed out and non-functional

## Specifications

version = 2.1.2 (4b29f7a2)
Architecture = ('64bit', 'WindowsPE')
Machine = AMD64
Platform name = Windows-10-10.0.18362-SP0
Processor = Intel64 Family 6 Model 58 Stepping 9, GenuineIntel
Python Branch = tags/v3.7.5
Python Build = ('tags/v3.7.5:5c02a39a0b', 'Oct 15 2019 00:11:34')
Python Compiler = MSC v.1916 64 bit (AMD64)
Python Implementation = CPython
Python Revision = 5c02a39a0b
Python version = 3.7.5
Release = 10
System = Windows
Version = 10.0.18362
Win32 = ('10', '10.0.18362', 'SP0', '')

  - Connected devices or roasting machine: Phidget TMP1101
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2020-02-26:
This is likely not changed in future versions as Artisan expects all events but for COOL to happen before DROP and being logged, or after, but not being logged. This makes internal handling of things simpler.

closed 2020-02-26 by MAKOMO

DennieSch, 2020-02-26:
Understand if that is the case. Just thought it might be straight forward to address since the SC Start and SC End buttons (while grayed out after Drop) were still functional (event was recorded) in V2.0.