artisan-roaster-scope/artisan #438: background image proportions in zoom

InvisibleOrdinary, 2019-12-05:
## Expected Behavior

background image retains same proportions when zoomed. It would be preferable that the background image remains locked in the same size and position as when in the zoomed-out / home view.

## Actual Behavior

proportions are stretched in odd ways

## Steps to Reproduce the Problem

  1. load a background image
  2. zoom to an area of the plot
 

## Specifications

  - Artisan Version: 2.1.1 (3c115a0) 
  - Platform (Mac/Windows/Linux + OS version): Mac 10.14.6 (18G1012)
  - Connected devices or roasting machine: non currently, but don't think that matters
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2019-12-22:
For technical reasons this is difficult to fix. Thus we decided to hide background images on zoom (Commit 8228f28c37b73bc7d388031b3807d14cc11c6d22)

closed 2019-12-22 by MAKOMO

InvisibleOrdinary, 2019-12-22:
That sounds good to me

Thanks

MAKOMO, 2019-12-22:
Thanks for this feedback and reporting the issue!