artisan-roaster-scope/artisan #439: glitches in graph drawing

InvisibleOrdinary, 2019-12-05:
I believe the sequence of events were (bu tnote I was roasting, so trying to move through this quickly):
 
Auto Charge triggered
It's didn't register quick enough, and I thought it didn't at all, so I triggered charge and started to adjust the charge position to where it actually happened. 
By doing that, I actually un-checked charge
I then re-checked charge at least twice trying to get in sync with the lag between me clicking the button and artisan registering what happened. (It was about here that I understood what happening)
I then moved the charge position to where it actually happened
While doing that, it glitched a couple times, and it took several attempts to get it in the correct position. 
By that I mean it would re-align the graph to the profile in the background (it's aligning to charge time) before it redrew it onscreen (maybe?) so that my cursor would be in the wrong position when I set the new charge position
That finally seemed to be correct, so continued roast
At end, I let FCe and drop go without registering in artisan until I hd the coffee in the cooling tray, and then went to set them manually.
While moving the cursor, it would show the graph back at about 1 min
you can see this behavior in the screen recording posted here:

https://www.dropbox.com/s/619usxedfos06tl/Screen%20Recording%202019-12-05%20at%2011.21.59%20AM.mov?dl=0

the roast was trashed, but the alog is uploaded here so you can see what happened.
 
[DEFT_SampleEval_373_19-12-05_1112.alog.zip](https://github.com/artisan-roaster-scope/artisan/files/3928716/DEFT_SampleEval_373_19-12-05_1112.alog.zip)

## Specifications

  - Artisan Version:2.1.1 (3c115a0) 
  - Platform (Mac/Windows/Linux + OS version): MAC 10.14.6 (18G1012)
  - Connected devices or roasting machine: phidgets: 3 temp probes, Barometric, humidity + ambient temp, ± 7kPa pressure sensor on gas
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2019-12-09:
Thanks Dave for this report. Hard to reproduce, but I saw the error message from the symbolic formula evaluation in the message line of your video. That formula asked for an index in the background data that did not exists and failed. I made a commit to catch this case and treat it better. In the best case this unhandled internal exception was the cause for the troubles you report here. The current continuous builds contain that fix. Please check if this resolves this issue.

InvisibleOrdinary, 2019-12-25:
I am not sure this is coming up for the same reasons, but I just witnessed it again in the current version (Artisan 2.1.2 (e5e2c5d) )

I didn't move quick enough to record the glitch. 
best I can relate is that it was just after I hit DROP
and then moving the mouse around made the last 1-2 minutes glitch in and out
seemed to be still recording
could trigger it once by clicking COOL but then it didn't do it again when unchecking and re-checking that
I will post more if I can recall anything else.

Thanks and happy holidays

Dave

MAKOMO, 2019-12-25:
Thanks Dave for this additional information point. I hoped to have resolved this by adding some protective measures in v2.1.2. I assume that something goes wrong, most likely triggered by adding an event, and from that point on an assumption of the incremental/caching redraw proves is not fulfilled anymore. The outdated (1-2 minute old) data you see alternating is from the bitblit cache that Artisan is using to speed up the otherwise too slow redraw. So Artisan makes kind of a screenshot and just draws new data on top of this. Somehow this process gets broken sometimes. Strange that this seems to happen rather rarely. I never observed this myself. I'll take another look especially at this DROP event implementation...

InvisibleOrdinary, 2019-12-25:
Okay. Sounds good.
The couple times I’ve seen it have been the ones I’ve reported, so I’m not seeing it often, but maybe there are time it would have been triggered and I didn’t do the ‘right’ thing.
Anyway, I’ll let you know if I can add anything more here.

Cheers

Dave

MAKOMO, 2020-01-25:
Just had the same effect with the current trunk. It happened after I entered an event (DE) using the corresponding event button and then repositioning it using a right-mouse click. Moving the mouse after this displayed the "old" cached canvas. Waiting for the next screen refresh corrected this until the mouse moved again. There seems to be an interaction between Artisans canvas bitblit caching and the one of the underlying matplotlib graphic lib. Might as well be a bug in matplotlib. I'll keep an eye on this now that I have an idea how to reproduce.

MAKOMO, 2020-02-25:
This one seems to be related to updating an event using the right-click popup menu while recording and an interaction with the draggable annotations mechanism, both of which use different bitblit caches that can conflict. This should be resolved be deactivating the draggable annotations while the events popup is operating in Commit f417c409b52c124ac187c3e24e0aeea24ef17af1.

InvisibleOrdinary, 2020-02-25:
Ok no will download and try to verify

Thanks

Dave

InvisibleOrdinary, 2020-02-26:
I ran though only a couple roasts this evening using build a57b574 (what was current by the time I went to download it). Didn’t see any glitches, and over all, I’d say even a little smoother. I’ll let you know if I see it crop up again.

Dave

MAKOMO, 2020-02-26:
Dear Dave, thanks for this quick test. Please download again the current version as the internal change on how to handle redraws broke the flavor chart, roasting report and wheel graph views. Something I just fixed with todays commit. This current version should be quite stable and you should be able to use it for production. Still we are only at the begin of the development of the next release version.

I close this issue for now, but will observe that issue and potentially reopen it if the issue re-appears.

Thanks,
Marko

closed 2020-02-26 by MAKOMO

InvisibleOrdinary, 2020-02-26:
Okay, that sounds good.

Thank you

Dave