artisan-roaster-scope/artisan #440: raspberry pi phidget22 driver not updated?

samtsai5212, 2019-12-07:
## Expected Behavior
When I compile the phidget22 drivers, it looks like the version should be 1.5
I also add phidget driver repository by the following way
https://wbcoffee.blogspot.com/2018/03/install-phidgets22-drivers-on-raspberry.html

## Actual Behavior
It shows phidgets22 ver 1.0.0 built April 8, 2018 which seem not updated.
[image]

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version:2.1.1
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version):
  - Connected devices or roasting machine:
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2019-12-10:
Unclear. Did you check again after a reboot?

MAKOMO, 2020-02-26:
stall

closed 2020-02-26 by MAKOMO