artisan-roaster-scope/artisan #449: File->Open/Save or Help->Save Settings crashes the program on Linux

DmitryBurstein, 2019-12-31:
## Expected Behavior
Open the appropriate file dialog

## Actual Behavior
Crash with Seg. Fault

## Steps to Reproduce the Problem

  1. File/Open

## Specifications

  - Artisan Version: 2.1.2
  - Artisan Build (number in brackets shown in the about box): 2d7c995
  - Platform (Mac/Windows/Linux + OS version): Linux Fedora 31
  - Connected devices or roasting machine: None
  
[artisan-settings-desktop.txt](https://github.com/artisan-roaster-scope/artisan/files/4011748/artisan-settings-desktop.txt)

MAKOMO, 2020-01-01:
Does v2.1.1 also show this issue on your setup?

DmitryBurstein, 2020-01-01:
I haven't tried 2.1.1, but 2.0 did also behave the same way.
I can open any saved profile by double-clicking in file browser, but not via the File menu of Artisan itself. Neither can I save one.

MAKOMO, 2020-02-26:
stall

DmitryBurstein, 2020-05-24:
The issue is still present: just tried the 2.1.3 pre-release, and it too seg-faults on file open.

MAKOMO, 2020-05-26:
Dmitry, I installed Fedora 31 on a VM and could duplicate your issue. Spend most of today trying to fix it and potentially catched this. Could you please give the current continuous builds a quick test?

https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous

Thanks,
Marko

DmitryBurstein, 2020-05-26:
Fantastic! Thanks a lot, everything works again. Forgot to mention I'm on F32 by now, but you can think of it as tested on their last version :)

MAKOMO, 2020-05-26:
Thanks for this confirmation

closed 2020-05-26 by MAKOMO