artisan-roaster-scope/artisan #450: Feature request: have Designer calculate milestones to create steadily declining rate of rise

drgarys, 2020-01-09:
To set roasting milestones in Designer, I can import the milestones for a roast profile. Moving the milestone points and changing their values makes the delta ROR curve move in predictable ways that are hard to adjust by moving points or changing their timing or temperate in the table that's available. I would love a feature where Artisan Designer would adjust the time signatures of milestones I entered to coincide with the steadily declining ROR that Scott Rao suggests. The ability to calculate other types of delta ROR might be useful also. 

I am using Artisan 2.1.2 on a 2015 MacBook Air running IOS 10.14.6, but I am really asking for a cross-platform feature that is not available in the latest version of Artisan.

roasterdave, 2020-06-08:
Thanks for the suggestion.  The Designer works only for creating BT curves from which the RoR is calculated, not the other way around.  It is possible to get a similar result another way.  Here is a procedure that makes use of the Analyzer to the create a curve with a straight line descending RoR for a given set of roast events. 

1. Open a profile (could be a real roast or one created in the Designer)
2. Run Analyzer>> Fit to x^^2 (in some cases it may require using x^^3 to get a reasonable fit)
3. Now use the menu Roast>> Switch Profiles, reply "Discard" when asked if you want to save the profile.

At this point the BT curve displayed fits the original roast event times and bean temperatures and results in a straight line declining BT RoR for the most critical part of the roast.  Notice the BT curve is the shape that corresponds to a fluid bed roaster.  Not represented is the initial temperature drop from CHARGE through TP and up to DRY END typical of a drum roaster.  When you think about it, the true temperature of the beans starts at the ambient temperature and rises continuously just as shown in this BT curve.  While the shape of the BT curve from CHARGE to TP to DRY could be manually recreated in the Designer it is probably not worth the trouble.  If adjustments do need to be made, take the curve into the Designer or use the Transposer (new in Artisan v2.4) to adjust the timings and then repeat steps 1-3 if necessary.

closed 2020-06-08 by roasterdave

drgarys, 2020-06-08:
Dave,

Thank you for a solution to what I’m trying to accomplish here. I’m going to save this procedure and work with it.

Gary

roasterdave, 2020-06-08:
Gary,

Thanks again for the question.  I'm sure others will benefit from this as well so I am preparing a post to the Artisan blog.  I'll come back and post the link here once it is published.

Dave

roasterdave, 2020-06-26:
Blog post: https://artisan-roasterscope.blogspot.com/2020/06/how-to-create-perfect-background-curve.html

drgarys, 2020-06-26:
Nice blog post, Dave! To join my exit from Dry to Ramp, how does Artisan
provide a BT ROR readout at that or any other point?

roasterdave, 2020-06-27:
Thanks drgarys.  Is your question about how to see the BT RoR while roasting or when looking at a static profile?

drgarys, 2020-06-27:
When looking at a static profile. BT ROR is displayed, of course, when a
roast is underway.

MAKOMO, 2020-06-27:
Dear drgarys, I have difficulties to understand your question. Could you elaborate a bit on this and maybe provide a picture showing your issue? The BT RoR is displayed on static profiles, if you ticked the corresponding flag, and also during recording after a CHARGE event is placed at any time point of the profile, including the "exit from Dry to Ramp". What exactly is the issue?

drgarys, 2020-06-27:
Hi Marko:

Thanks for responding. The solution by Dave to create a background profile with optimal delta ROR is terrific. It displays for a fluid bed roaster, where the curve is a constant rise in temperature, with no turning point. I can go into the Designer curve created with his method and see the delta BT at End of Dry. That would then be my target

I am trying to tie together the curve I get on my drum roaster with the Designer curve that proceeds past End of Dry. If I go to the Roast Properties, Data tab, I can see a delta BT ROR. But it varies a lot between readings. Here is a picture.

To steer to a target ROR when I hit End of Dry I could manually average from 3:30 to 3:45, which would give me a delta BT ROR of 203.6/6=33.93 (about 34). If my target ROR is different, I need to adjust my roasting procedure to reach End of Dry at the target setting. 

I think I have my own answer here. I will need to go in and do my manual averaging to predict how my procedure on my roaster gets me the target ROR needed at End of Dry. (Practice, practice.)

Thanks again,
Gary

MAKOMO, 2020-06-29:
Gary, there was no picture attached (here you can attach  an image just by drag and drop; select the "Preview tab" to verify). If you need to get the RoR of the chart, why not just observe the mouse curser widget that gives you the coordinates (x and y-coordinates) of the mouse. The x-coordinate is the time and the y-coordinate is by default it reports the position on the temperature axis, but pressing the d-key toggles that to the RoR axis.

[image]

drgarys, 2020-06-29:
Marko:

I hope you get the image now. My reasoning of not choosing a single point is that my BT ROR reading is so variable that an average of a few points is more useful.

Gary

MAKOMO, 2020-06-29:
I don't see an image here, but for mine. Do you see one? If you BT Delta is "so variable" after you tuned the RoR smoothing under Config >> Curves, you might have a fundamental noise issue. Check my [On Idle Noise](https://artisan-roasterscope.blogspot.com/2019/03/on-idle-noise.html) post and improve your hardware setup.

drgarys, 2020-06-29:
Marko:

Here’s the image as an attachment. Please let me know if this looks too noisy. Then I’ll follow up on the idle noise adjustment.

Thanks,
Gary

MAKOMO, 2020-06-29:
There is just  nothing attached. Could you please browse to the Web page of this issue at

https://github.com/artisan-roaster-scope/artisan/issues/450

to verify? There you can just drag-and-drop your picture, but not by replying to an email towards this system.

Thanks,
Mario

drgarys, 2020-06-29:
Marko:

I didn't realize I couldn't communicate via email alone. Now I'm signed in. Here's the image, finally.
[image]

MAKOMO, 2020-06-29:
The data in this table shows the raw RoR values without any smoothing applied. Raw RoR values are amplifying any noise of the underlying data (BT and ET in this case). The amount of noise on the ET/BT curves depend on your measuring setup and if that is noisy your raw RoR readings will be dramatically noisy. As said before. It might be better for you to pick the RoR from the chart using the mouse pointer and the xy-coordinate display as this data applies the smoothing settings that you configured for the RoR curves and thus does the averaging that you are looking for. Maybe I still did not understand your issue well.

drgarys, 2020-06-29:
Thank you, Marko. That's what I needed to know -- that the mouse pointer displays ROR with smoothing settings I configured.