artisan-roaster-scope/artisan #451: Activating the control button on the Hottop

rmongiovi, 2020-01-26:
Now that the UI has been changed to hide the time LCD until Artisan is turned on, is there any chance it could remove that requirement to right-click on the LCD to activate the Control button?  I'm not sure what that right-click was really supposed to accomplish since you already have to configure the UI to support the Control button but it's a bit inconvenient to have it based on an element that's not present in the UI until you press the On button.

I have my alarms configured to set the power and fan after On.  So my workflow used to be right-click the LCD, press control, turn On.  Now it has to be turn On to bring up the LCD, right-click the LCD, turn Off so that I can turn On again, turn On with Control enabled.  If I turn On without control being enabled my first two alarms don't function.  I tried setting a 30 second Time on them to give me a chance to enable Control after On but it didn't seem to work.  I guess I don't know how to use the Time column.

In any case, that whole right-click thing just seems unnecessary to me.  I specifically configured Control in the device configuration menu to provide the button.  I press the Control button and the Hottop drum starts spinning, I know Control is enabled.  I read your explanation about "commercial roasters have safety precautions" but this isn't a commercial roaster and I'm clueless as to what I'm being protected from.  All I can see is that it's protecting me from using Artisan to control the roaster which is exactly what I'm using Artisan to do.

roasterdave, 2020-01-27:
The default behavior did change but you can return to having the time LCD show in the OFF state.  Click on the menu item View>Readings.  You can now use your previous workflow.

rmongiovi, 2020-01-27:
Oh yeah?  Cool!  Thanks.

rmongiovi, 2020-01-29:
I think I do have to comment on the whole right-click secret handshake, though.

Right now I have to right-click on the Timer LCD to enable the Control button.  If I fail to do that and just click on the Control button I get a popup telling me I should do right-click.

Isn't the more normal way to have some sort of protection on a potentially dangerous button to have clicking on the button bring up a popup asking me if I meant to click?

Clicking on the Control button and then answering a popup that says "Enable Control (Yes/No)" seems more natural to me than a magic right-click on the Timer LCD....

MAKOMO, 2020-02-26:
thanks for your thoughts, but we currently do not intend to change this behavior

closed 2020-02-26 by MAKOMO