artisan-roaster-scope/artisan #455: Feature Request: Phidget DCC1002 Support

wesleymerrick, 2020-03-14:
I've recently started running my roaster's fans through the Phidget [DCC1002](https://www.phidgets.com/?tier=3&catid=18&pcid=15&prodid=1117) and would love to be able to adjust its setting directly from artisan. So far as I can tell the DCC1002 hasn't been implemented as of 2.1.2, but if I've missed something (especially in the unreleased code) please let me know.

I'd be glad to attempt writing the implementation myself, @MAKOMO do you all have any developer docs for how to build the project?

Thanks for maintaining this awesome tool for roasters!

MAKOMO, 2020-03-16:
This motor controller is not yet support. It has a rather rich API, actually 3 separate APIs, so full support is kind of "expensive" (besides that I have no module and motor to test).

Would it be enough to get going for you if setTargetVelocity() and setAcceleration() the DCMotor API would be supported?

There are no developer docs, besides the (complete) scripts to run and build Artisan that are in the repository. Sorry.

wesleymerrick, 2020-03-17:
Yes, l should have been more specific with regards to the features of the DCC1002. I don't use an encoder (don't care about the position of the fans) nor do I want to apply negative/braking voltage. Really just having a binding on setVelocity() within artisan is all I'm after, being able to use buttons or a slider rather than a separate app would be fantastic. If it's not too much of a stretch to support the DCC1000 as well, assuming they share the same API, that would be great.

I'll spend some more time looking through the build scripts and try to get something working on my end.

MAKOMO, 2020-03-17:
In Commit 673b5aaa841471ecd7eb0f78fa95e3976421d77b I added basic support for the Phidget DCMotor API supporting the VINT DCC1000, DCC1002 and DCC1003 modules. New builds should be available in about 30min at https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous

There are two new IO Commands defined

- `accel(c,v[,sn])` with `c` channel number and `v` acceleration as a float, and `sn` serial the optional serial/port number of the addressed module
- `vel(c,v[,sn])`   with `c` channel number and `v` target velocity as a float, and `sn` serial the optional serial/port number of the addressed module

For lack of hardware I couldn't test if this works at all. Could you please test on your hardware and report back?

wesleymerrick, 2020-03-18:
Thanks for working on this! I installed the continuous release version from a few hours ago, but this is my first time configuring an output device in artisan so I probably made a mistake somewhere, if not more than one. Here's what the Phidget control panel shows with my DCC1002 connected via the HUB0000: 
[image]

And here's what I set up for the slider under the events menu in artisan:
[image]

Starting a roast and moving that slider had no effect on the fans, but again I'm sure I missed something in configuring the output device. Do I need to make an entry in the Extra Devices tab of the Devices menu? If so what settings should I use there?

I did run the example python code from the Phidgets website which calls setTargetVelocity(1) after attaching the device and that did get the fans moving, so implementing that will definitely get the functionality I'm after.

Let me know what other configuration options I should try and if there's any more diagnostic information I can gather to help troubleshoot this.

MAKOMO, 2020-03-18:
Note that you need to keep the Phidget Control Panel and any other program accessing the phidget driver closed to allow Artisan to connect to the device. Could you try a simpler setup first: define an extra button with action "IO Command" and put in the description just "vel(0,1)" (no serial number or port). Note that those serial numbers are only needed if you want to operate several modules of one kind. Note further that the slider command as you defined it substitutes the placeholder {} by the values from interval 0-100. However, it seems that the setTargetVelocity() operation has a range of only -1 to 1. So you might want to set the factor of your slider to 0,01 instead of 1.

I added some debug messages to the message log (menu Help >> Messages). Could you please download the latest version from the same link, test and send me the content of the message log window to check where it hangs?

Thanks!

wesleymerrick, 2020-03-18:
Ok, I made sure the control panel was closed and added two buttons to artisan, Air100 with the vel(0,1) IO command and AirOFF with the vel(0,0) command. Here's a screenshot of the Messages log after pressing those buttons during a roast:
[image]

MAKOMO, 2020-03-19:
My bad. I introduced a typo in a message output command. Could you retry please with the new builds. Sorry for the ping-pong.

wesleymerrick, 2020-03-19:
No worries, I'm glad to help where I can, and being able to work across time zones is one of the great things about open source.

I ran the new build and it turned up a different error, below is another screenshot:
[image]

Thanks for all your work towards getting this implemented.

MAKOMO, 2020-03-20:
Excellent! Here it is the typo. I fixed it and removed the extra logging. Should work now as expected. Thanks for testing!

wesleymerrick, 2020-03-21:
That did it! The vel() function works with both my DCC1000 DCC1002. I don't have a means to really test the accel() since I don't have an encoder for the fans or anything.

I just finished a few roasts using some new buttons for the fan control and it's greatly simplified changing and recording my fan settings. Thanks for all the time and attention you gave this.

MAKOMO, 2020-03-21:
Thanks for supporting me on this by testing all those versions. Happy roasting!

closed 2020-03-21 by MAKOMO