artisan-roaster-scope/artisan #456: Serial Exception:Unable to open serial port Extech 755

HansGem, 2020-03-22:
## Expected Behavior
Added  supported manometer Extech HD755 to Ports Configuration in Extra tab, expecting Artisan to log gas pressure 

## Actual Behavior

No connection, but message "Serial Exception:Unable to open serial port"

## Steps to Reproduce the Problem

  1. Tried various number for ports beside the default 'COM1' that shows up
  2. Have tried to locate allocated port for the Extech, but although the Extech shows up when usb plugged in, location ID indicted makes no sense to me - where do I find which Comm port?- unable to find that via terminal or about this mac's system report. 
  3. Searched web last two weeks, no joy :-( 
[artisan-settings.txt](https://github.com/artisan-roaster-scope/artisan/files/4364757/artisan-settings.txt)

## Specifications

  - Artisan Version: 2.1.2
  - Artisan Build (number in brackets shown in the about box):e5e2c5d
  - Platform (Mac/Windows/Linux + OS version): Mac Catalina 10.15.3
  - Connected devices or roasting machine: Yoctopuce VirtualHub-1.10.38634; 
Yocto-Thermocouple and Yocto-Meteo-V2
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2020-03-22:
Did you install the CP210x driver from Silicon Labs that this device requires?

https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers

HansGem, 2020-03-22:
Tx Marco downloaded and installed, BUT, sadly still get same message, even after re-starting laptop and Artisan. Tried Comm0 through to 8. Same msg each time.

HansGem, 2020-03-22:
CP2102 USB to UART Bridge Controller:

  Product ID:	0xea60
  Vendor ID:	0x10c4  (Silicon Laboratories, Inc.)
  Version:	1.00
  Serial Number:	0001
  Speed:	Up to 12 Mb/s
  Manufacturer:	Silicon Labs
  Location ID:	0x14100000 / 8
  Current Available (mA):	500
  Current Required (mA):	100
  Extra Operating Current (mA):	0

MAKOMO, 2020-03-22:
You specified that you are working on macOS, but macOS does not address serial ports via Comm0,.. through 8, but via /dev/ devices. You need to set the port for this extra device in the Ports Configuration dialog (menu Config >> Ports) in the "Extras" tab, no the ET/BT tab. There should be a popup in the Comm Port column for this extra device that should be populated with the correct /dev/.. port automatically while the device is plugged in and the driver is loaded. Just click it while the device is disconnected and again while it is connected. The one additional menu entry should be the correct port.

HansGem, 2020-03-23:
Fantastic Marko: all works fine now. Didn't realise the drop-down/popup. 
Is there a way to multiply output ie .250psi to graph as 250?

MAKOMO, 2020-03-23:
Excellent! Could you please post the solution on Home-Barista that all benefit from this and your the thread there gets closed?

You can add a symbolic formula "x*100" to multiply the reading by 100 under Config >> Device, Extra Devices in the y1(x) field (without the double-quotes).

closed 2020-03-23 by MAKOMO