artisan-roaster-scope/artisan #458: The decimal place is one place out on LCDs. That is it shows 17degrees instead of 170deg?

krametoof, 2020-04-04:
## Expected Behavior

## Actual Behavior

## Steps to Reproduce the Problem

  1. Always there 
  2. 
  3. 

## Specifications

  - Artisan Version:
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version):
  - Connected devices or roasting machine:
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2020-04-04:
This report lacks all necessary information on your setup. Which Artisan version are you using, on which platform, which hardware is connected, what is your exact Artisan setup (as .aset file). Please add this information or I will have to close this issue. Thanks.

krametoof, 2020-04-04:
Yes sure. The operating system is windows 10 connecting to a dongyi dy-1ky. Ive just installed artisan 2.1.2.

MAKOMO, 2020-04-04:
The dongyi dy-1ky is not a supported machine. You need to ask the producer of this machine for support.

closed 2020-04-04 by MAKOMO

krametoof, 2020-04-04:
Unfortunately i dont imagine any help from them. Interestingly the lcds showed correctly on the earlier version of artisan

krametoof, 2020-04-04:
I'll ask a blog group for info. I imagine its in the settings and a simple change but i have not yet discovered it.