artisan-roaster-scope/artisan #459: 'Config|Curves...' window not resizable and doesn't respect display size

DennieSch, 2020-04-14:
## Expected Behavior
Upon opening the 'Config|Curves...' window, it would (re)size itself so it is no larger that the display size. This would allow it to display the 'Ok' and 'Cancel' buttons. If they were 'off screen', ability to resize the window in order to see/display these button. 

## Actual Behavior
Window is sized to beyond full vertical size thereby preventing display and access to the 'Ok' and 'Cancel' buttons. Also, while there are resize handles to appear on the top and left side (none on the right side) they do not allow resizing of the window.

(see full display screen capture attached - a jpg with an added .zip)
[Curves window oversized.JPG.zip](https://github.com/artisan-roaster-scope/artisan/files/4476409/Curves.window.oversized.JPG.zip)
My device settings -- 
[my device display settings.JPG.zip](https://github.com/artisan-roaster-scope/artisan/files/4476582/my.device.display.settings.JPG.zip)

## Steps to Reproduce the Problem

  1. Click 'Config|Curves...'
  2. 
  3. 

## Specifications

version = 2.1.3 (a65212c0)
Architecture = ('64bit', 'WindowsPE')
Machine = AMD64
Platform name = Windows-10-10.0.18362-SP0
Processor = Intel64 Family 6 Model 58 Stepping 9, GenuineIntel
Python Branch = tags/v3.8.0
Python Build = ('tags/v3.8.0:fa919fd', 'Oct 14 2019 19:37:50')
Python Compiler = MSC v.1916 64 bit (AMD64)
Python Implementation = CPython
Python Revision = fa919fd
Python version = 3.8.0
Release = 10
System = Windows
Version = 10.0.18362
Win32 = ('10', '10.0.18362', 'SP0', '')

  - Connected devices or roasting machine: N/A
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

roasterdave, 2020-05-22:
I was not able to duplicate this on my Windows 10 machine which has the same resolution and display settings as yours.  Would you post a screenshot of the UI tab, which sets the height of the dialog, so we can see how that appears?

DennieSch, 2020-05-22:
Hey Dave - Attached is a 'Desktop' screen capture with the UI tab selected. 
[UI Screen Capture.zip](https://github.com/artisan-roaster-scope/artisan/files/4669976/UI.Screen.Capture.zip)

roasterdave, 2020-05-22:
Thanks Dennie.  That helps.  Still can't reproduce it, nevertheless I will shrink the size of the dialog so it will fit on smaller screens.

DennieSch, 2020-05-22:
Dave- I think I've found something!
Here is a screen capture of the Windows 'Display' settings screen. Note the Scale and Layout section. I've been operating with the Recommended 150% setting. At that setting, the apps on my machine are very readable readable (don't need a magnifying glass to read the the text). If I change this to 125% or 100% then... I can see the entire Curves window. However, while technically readable the fonts are quite small. 
See attached - 
[Display - Scale and layout.zip](https://github.com/artisan-roaster-scope/artisan/files/4670426/Display.-.Scale.and.layout.zip)

roasterdave, 2020-05-22:
Dennie, that is a good answer for you for now until the dialog shrinks.  Then you will be able to go back to 150%.  Thanks for the update!

MAKOMO, 2020-05-23:
Dennie, I reduced the size of that dialog further and changed (improved?) the way it handels/remembers the dialogs size and position. Hope this helps further. Thanks for reporting.

DennieSch, 2020-05-23:
Marko, It looks good and works great. Thank you!

closed 2020-05-25 by MAKOMO