artisan-roaster-scope/artisan #460: Hanging on open on MacOS Catalina 10.15.4

squarefeetcoffee, 2020-04-21:
## Expected Behavior

## Actual Behavior

Upon opening, the app doesn't respond.  I have tried this on two different Mac computers (2019 Mac Mini and a 2017 iMac).  Both Macs are running OSX Catalina 10.15.4.

I'd be happy to help with any troubleshooting as well as share my computer via TeamViewer or similar.

## Steps to Reproduce the Problem

  1. Installed latest Artisan Scope, downloaded April 21
  2. Open the App
  3. After 5 minutes, it is still not open and when I select Option-Command-ESC, it reports that it is hung.

## Specifications

  - Artisan Version:
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version):
  - Connected devices or roasting machine:
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2020-04-21:
Did you download the release version v2.1.2 or a development build!? Are you running Artisan in full screen mode?

squarefeetcoffee, 2020-04-21:
Hi Marko,

2.1.3, I’m sorry, I should have made that more clear.  It has been quite a few years since I wrote code :-)

Yes, I’m in full screen by default. As it is hung, I am unable to get out of full screen…

—Dave

squarefeetcoffee, 2020-04-21:
I can't find the version as the app is 100% locked, but it is 2.1.3.  I just downloaded it again and have the same issue, so whatever version is currently the Continuous Build is what I have.

I also tried removing all of the settings (I unmounted the disk with my Artisan files in it), as well as deleted the artisan-scope folder in ~/Library/Application Support.

MAKOMO, 2020-04-21:
Ok. I will take a look as soon as I find the time. Please use the release version v2.1.2 until this is resolved.

https://github.com/artisan-roaster-scope/artisan/releases/tag/v2.1.2

squarefeetcoffee, 2020-04-21:
If you need me to do anything from my end, I’d be happy to.  I have Team Viewer on the Mac, so I can also easily share it.

MAKOMO, 2020-04-23:
Could you please give the current v2.1.3 development build a test and report if this resolves the issue for you? Thanks!

squarefeetcoffee, 2020-04-23:
Marko, it appears to work perfectly!  Thank you!!!  —Dave

MAKOMO, 2020-04-23:
Thanks for testing and this confirmation!

closed 2020-04-23 by MAKOMO