artisan-roaster-scope/artisan #461: Error on FC START / FC END / SC START

splenguin, 2020-04-29:
## Expected Behavior
Click button for FC START / FC END / SC START, record events

## Actual Behavior
ERROR:
File "artisanlib/main.py", line 8670, in mark2Cstart

--------------------------------------------------------------------------------
artisanlib/main.py::mark2Cstart:8670
	                self = <artisanlib.main.tgraphcanvas object at 0x71b4cdf0>
	             removed = False
	                   _ = False

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version: 2.1.2
  - Artisan Build (number in brackets shown in the about box): 2d7c995
  - Platform (Mac/Windows/Linux + OS version): Raspbian
  - Connected devices or roasting machine: Phidget 1048 [4xTC]
[artisan-settings.txt](https://github.com/artisan-roaster-scope/artisan/files/4552611/artisan-settings.txt)

  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2020-04-29:
Thanks for the report. I cannot duplicate this one (on my mac, that should not be much different to the RPi in this respect) and the error message seems to lack the lines detailing the actual error. What are the exact steps to reproduce this one? Start Artisan with your settings, press START and then press the buttons FC START, FC END,  and SC START in this order, without pressing CHARGE before? Any more information that you could share to help to iron this one out?

splenguin, 2020-04-29:
Hi Makomo, thanks for responding. I have a little more info, but yes, start Artisan, click On, click Start. What I didn't notice previously is that it actually throws an error when I click CHARGE, just not a pop-up error (just a line of text above the plot area):
1 13:52:26.862 Error: File "artisanlib/main.py", line 8492, in mark1Cstart -------------------------------------------------------------------------------- artisanlib/main.py::mark1Cstart:8492 self = removed = False _ = False 
So then when I slick FC START, it pops up the error I mentioned in my original post. Seems weird that it's referencing mark1Cstart when using the CHARGE button... does that help at all?

MAKOMO, 2020-04-29:
Well to me this looks like you are running Artisan v2.1.2 on an unsupported platform. Are you on Raspian Stretch? Note that Buster is not explicitly supported as it is untested. I am aware that Buster runs a very different Qt version and that might just be incompatible to the messaging assumed by the one referred to by Artisan. I have currently no RPi around to test this. Sorry.

MAKOMO, 2020-04-30:
Found my RPi and it is running under a clean installed Raspian Buster. Artisan v2.1.2 works fine, including those event buttons. I cannot reproduce your issue. Any more information available on this?

splenguin, 2020-04-30:
I am running Buster, but will switch to Stretch since this is a single-purpose Pi. I did a roast last night. On first launch Artisan was still problematic, I noticed also that the UI was different (not showing LCDs) and it wasn't receiving data from the Phidget. HOWEVER, after messing around (and not changing any config) I restarted Artisan and it was instantly back to normal, and I was able to do the roast without issue, including events. So possibly I'm just having a stability issue on Buster that Stretch would resolve.

MAKOMO, 2020-04-30:
Wait a second. Let me try to make a build for Buster (using a more modern GUI lib setup which might have advantages on its own) on my setup for you to test. If that works out I can publish a Buster build next to the standard Stretch build that is (has to) be based on quite outdated GUI libs.

splenguin, 2020-04-30:
Happily will!

MAKOMO, 2020-05-04:
By now I drafted a test build that should run on Buster and contains many updated libs. 

https://github.com/artisan-roaster-scope/artisan/releases/tag/v2.1.3-beta

Please report if this is working for you. Thanks!

splenguin, 2020-05-07:
Ok I've got 2.1.3 installed, so far so good. I will let you know as soon as
I do a roast with it. Thank you very much!

MAKOMO, 2020-05-07:
So it installs and starts up? Already a promising result.

splenguin, 2020-05-07:
Sure does!

splenguin, 2020-05-16:
So I've done two roasts now with 2.1.3. I've had a few weird UI issues in
either the RoR not plotting until the roast is done, or the event buttons
at the bottom missing. I have not seen the issue again that I initially
described in this ticket. Also I've seen the missing RoR in 2.1.2 as well.
So far I've been able to get both issues resolved just by restarting the
application or the device, etc. So it is fully functioning, just with a
little messing around at setup. I'm happy to provide a log or whatever's
appropriate next time if you'd like, you'd just have to tell me how to
collect it.

MAKOMO, 2020-05-16:
The RoR curves are drawn on recording only after the CHARGE event is set. The event buttons can be shown or hidden using the corresponding entry in the View menu (or via its keyboard shortcut key b). Artisan remembers the visibility of UI elements per state (off, on, recording). Hope this helps. Note that there is an updated build of v2.1.3 now available with is the Release Candidate for the upcoming release v2.4 that is to be expected to be released later this month.  https://github.com/artisan-roaster-scope/artisan/releases/tag/v2.4.0-RC

closed 2020-05-16 by MAKOMO