artisan-roaster-scope/artisan #462: Alarm Pop Up window

DennieSch, 2020-05-03:
## Expected Behavior
Alarm Pop Up window is displayed both in the app and WEBLCDs.

## Actual Behavior
Alarm Pop Up window... no longer 'popping up'. Pop up is no longer popping up on the WEBLCDs either.  It use to be working just fine. I have not changed anything in my 'Alarms page' for many months. 

## Steps to Reproduce the Problem

  1. Alarm ON is checked, Action is set to Pop up, Beep is also checked.
  2. Start a roast
  3. Beep does happen but no pop up in app nor in WEBLCD. Call Program does work. 

## Specifications
I'm frequently updating my install with the Continuous Builds. I THINK it was working when I roasted on 4/15/2020.

version = 2.1.3 (95cfc9e5)
Architecture = ('64bit', 'WindowsPE')
Machine = AMD64
Platform name = Windows-10-10.0.18362-SP0
Processor = Intel64 Family 6 Model 58 Stepping 9, GenuineIntel
Python Branch = tags/v3.8.0
Python Build = ('tags/v3.8.0:fa919fd', 'Oct 14 2019 19:37:50')
Python Compiler = MSC v.1916 64 bit (AMD64)
Python Implementation = CPython
Python Revision = fa919fd
Python version = 3.8.0
Release = 10
System = Windows
Version = 10.0.18362
Win32 = ('10', '10.0.18362', 'SP0', '')

  - Connected devices or roasting machine: N/A
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2020-05-03:
Dear Dennis thanks for this detailed report. Fixed in source but not commited yet.

A note on the continuous builds: 

Good for us that you are testing those and report back. This helps a lot as for example this issue would not have been discovered by us for a while. Bad for you as things might just not work as expected. Very risky for your roasts!

Those builds are pure development states and might not work at all. Often they contain changes that are completely untested. Currently we are, for the first time ever, applying some large refactoring touching almost every line and potentially breaking every feature. As this is a hugh task this is done in steps. Thus do not use any of those builds in the next 5-7 days until things stabilised somewhat. We plan to soon release a stable Release Candidate for intensive community testing before releasing the regular new version.

closed 2020-05-03 by MAKOMO

DennieSch, 2020-05-03:
Hi Marko- No problem, glad to help.
I do understand that those builds are, 'use at your own risk'. I usually keep a couple of the previous builds/commits on hand. That way, if the 'wheels come off' I can roll-back to an earlier commit.
Just looking to provide feedback and suggestions to help you and the team make the best tool.
Cheers - Dennie