artisan-roaster-scope/artisan #463: 'Tools|Comparator...' display of selected roast information

DennieSch, 2020-05-04:
## Expected Behavior
To be able to display selected information of a roast of those being compared.
It would be very helpful to be able to display information for a particular roast when using the Comparator in order to analyze differences in roasts. 

For Example --
Right Click (PC) on a roast title
Right Click would bring up a short menu consisting of -- 

**Properties** (information in the top portion of a roast report)
**Events** 
**Rst Notes**
**Cupping Notes**

Upon selection of one of the above 4 options, a window would pop up displaying that information.

When comparing roasts, this would allow one to drill down into the roast information to help explain and better understand differences in the roasts being compared,.

## Actual Behavior
Not currently available - feature request

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications

version = 2.1.3 (b7411d3b)
Architecture = ('64bit', 'WindowsPE')
Machine = AMD64
Platform name = Windows-10-10.0.18362-SP0
Processor = Intel64 Family 6 Model 58 Stepping 9, GenuineIntel
Python Branch = tags/v3.8.0
Python Build = ('tags/v3.8.0:fa919fd', 'Oct 14 2019 19:37:50')
Python Compiler = MSC v.1916 64 bit (AMD64)
Python Implementation = CPython
Python Revision = fa919fd
Python version = 3.8.0
Release = 10
System = Windows
Version = 10.0.18362
Win32 = ('10', '10.0.18362', 'SP0', '')

  - Connected devices or roasting machine: - N/A
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2020-05-19:
A click on a row label opens the corresponding profile in the ArtisanViewer if running providing all information in all detail. Thus we feel only a very weak need for duplicating this data in other places. Thanks for this suggestion, but we went another route.

closed 2020-05-19 by MAKOMO

MAKOMO, 2020-05-24:
Dear Dennie, I picked up your idea somewhat and just added tooltips showing some of a profiles metadata on hovering over a profiles title in the Comparator tool window.

Thanks for the hint!

[image]

DennieSch, 2020-05-24:
Nice... works well. I think this will be helpful. 
I did notice that the following (the double click feature) I mentioned in #465 is still not working for me (on a Windows 10 machine). 

*Comparator
Enhancement -- It would be nice to be able to Align - on TP also.

Not working --
Description says... "A double click on a profile row header opens the corresponding profile in a second Artisan instance running in Viewer mode for further inspection."

Double click
~5s later; Comparator window title bar changes from "Comparator" to "Comparator (Not Responding)"
after another ~5s; Comparator window is moved to the background, clicking on the main Artisan window brings the Comparator window to the foreground (Always On Top). No second instance of Artisan (Viewer mode) is started.

roasterdave, 2020-05-26:
Dennie, regarding the Comparator (Not Responding), is it possible you have a hanging Artisan process?  The best way to check is to exit all Artisan instances and then use Windows Task Manager to see if there is still an Artisan instance showing.  Terminate it if there is one.  Then retry the Comparator, but first open a second instance of Artisan to start the ArtisanViewer.  ArtisanViewer will not start automatically start when double clicking in the Comparator.  It must already be running to receive and display the profile.

DennieSch, 2020-05-26:
Dave, thanks for the info but no difference in behavior. 

Steps performed -- 
Restarted computer
Launch Artisan
Launch second instance of Artisan
(task bar shows two instances; "Artisan 2.1.3" and "ArtisanViewer 2.1.3")
Using main/first Artisan instance, start Comparator 
Selected 3 .alog files to compare - 3 roasts are displayed
Double click on row header (any of the 3)
Same behavior as noted earlier.
Nothing is displayed in ArtiasanViewer
Task Manager shows 2 instances of Artisan.exe are running

Anything else you'd like me to try to help?
Dennie

roasterdave, 2020-05-26:
Dennie, Thanks for checking that.  This feature is working fine on my Windows machines so we will get there on yours!  

Something else to try please.  This will confirm if the interapp communication is functioning.  With both Artisan and ArtisanViewer running please double click on an alog file in the Windows Explorer.  Does it open in the Artisan?  Now put Artisan into Designer mode.  Again double click the alog file in the Explorer.  Does it open in ArtisanViewer?  

Thanks,
Dave

DennieSch, 2020-05-27:
Dave; Got a hard fault.

"With both Artisan and ArtisanViewer running please double click on an alog file in the Windows Explorer."
Result - alog loads in Artisan window

"Now put Artisan into Designer mode."
Result - with the alog loaded in previous step, selected Designer mode and program error occurs. 

Alternate test.
With both Artisan and ArtisanViewer running.
Placed Artisan in Designer mode (without loading an alog file) - Artisan enters Designer mode.
Double clicked on an alog file in the Windows Explorer.
Result - behavior is the same as that seen with the double click in Comparator. After ~5s, Artisan window title bar shows "Artisan 2.1.3 (Not Responding)" for ~10s and after which, it is then available again.

Files attached.
[Error and Additional Info.zip](https://github.com/artisan-roaster-scope/artisan/files/4686194/Error.and.Additional.Info.zip)

roasterdave, 2020-05-27:
Dennie, Thanks for posting this additional information.  The settings file held the key.  Will you please turn off Web LCDs and then try to open a file in ArtisanViewer by double clicking in the Comparator?  With confirmation this works on your system we can sort out a permanent solution.
Dave

DennieSch, 2020-05-27:
Dave, That worked.
With Web LCDs off, double clicking in Comparator did load that profile in ArtisanViewer. No 'delays' or faults.
Dennie

roasterdave, 2020-05-27:
Dennie, thanks for confirming.  Will be fixed.
Dave

roasterdave, 2020-05-27:
Dennie,  the profile you shared generates a fault in the Designer because the SCe event occurs after DROP.  That is not an issue normally.  However, the Designer expects and needs the roast events to be in order.   The fault in the designer will get trapped in a future build but the Designer still will not be able to handle a profile with the events out of order.  If you ever want to import that profile into the Designer the event times can be edited at the top of the Events tab of the Roast Properties dialog. 

Thanks for coming across this one!!
Dave

DennieSch, 2020-05-27:
Dave, Got it on the reason for the fault. Sounds good. 

I had brought up the question of recording SCe (and SCs) after Drop in #437 . There I describe how my roasts typically go towards the end. What I've been doing is writing down SCe timing and while the roast is cooling, I go into Events and manually enter the SCe time. I'm trying to record/retain the duration of SC even if it started and/or ends after my Drop/Cool begins. 

Thanks,
Dennie

roasterdave, 2020-05-28:
Dennie, there are some updates that would be good for you to try.  You can reenable the WebLCDs.  They no longer interfere with double clicking in the Comparator to open a profile in ArtisanViewer.  Also, ArtisanViewer will now autostart if it is not already running.

Thanks for testing!
Dave