artisan-roaster-scope/artisan #465: Ranking Report (Web) error message

DennieSch, 2020-05-12:
## Expected Behavior
No error message

## Actual Behavior
Report generated with error message

## Steps to Reproduce the Problem

  1. File|Report|Ranking|Web...
  2. Selected one file
  3. Error message generated --

3 23:49:53.200 Exception: rankingReport() 'int' object is not subscriptable@line 28497
2 23:49:51.951 Scope has been reset
1 23:49:32.816 

## Specifications

version = 2.1.3 (226bab11)
Architecture = ('64bit', 'WindowsPE')
Machine = AMD64
Platform name = Windows-10-10.0.18362-SP0
Processor = Intel64 Family 6 Model 58 Stepping 9, GenuineIntel
Python Branch = tags/v3.8.0
Python Build = ('tags/v3.8.0:fa919fd', 'Oct 14 2019 19:37:50')
Python Compiler = MSC v.1916 64 bit (AMD64)
Python Implementation = CPython
Python Revision = fa919fd
Python version = 3.8.0
Release = 10
System = Windows
Version = 10.0.18362
Win32 = ('10', '10.0.18362', 'SP0', '')

  - Connected devices or roasting machine: N/A
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2020-05-12:
Dear Dennie, thanks for this report. Good catch. A typo introduced while doing a major restructuring. Fix with the next commit. We count on your testing of the official Release Candidate to be released later this week, hopefully with some documentation of the changes included. There might be more typos like this one still hiding....

closed 2020-05-12 by MAKOMO

MAKOMO, 2020-05-16:
Dear Dennie, I just uploaded the Release Candidate for the next Artisan v2.4 (still reporting v2.1.3). Some documentation will published later today or tomorrow.

https://github.com/artisan-roaster-scope/artisan/releases/tag/v2.4.0-RC

DennieSch, 2020-05-19:
Hello Marko - Not sure if posting this here is the proper venue but thought this might be easier for now given this is a collection of items.

Artisan RC 2.4.0 (version = 2.1.3 (532987c5))

*Comparator
Enhancement -- It would be nice to be able to Align - on TP also.

Not working --
Description says... "A double click on a profile row header opens the corresponding profile in a second Artisan instance running in Viewer mode for further inspection."

Double click
~5s later; Comparator window title bar changes from "Comparator" to "Comparator (Not Responding)"
after another ~5s; Comparator window is moved to the background, clicking on the main Artisan window brings the Comparator window to the foreground (Always On Top). No second instance of Artisan (Viewer mode) is started. 

*Simulator
Auto saving occurring when 'stopping' the simulation via clicking 'OFF'
Standard event buttons (charge, dry, FCs, etc.) and sliders are active and get recorded
No deltas graphed
No events with their annotations are being displayed (all events in properties are at 00:00)
No slider events are being displayed
Is there a way to 'pause' a simulation??
"Scope recording..." vs. roast title (of the roast being simulated) is displayed in the message line

*Autosave
I've been using the new autosave shortly after it was introduced in an earlier commit.
My 'prefix' is  ~titleC_~weight~weightunits_~datetime_~batch_long
An example of the result for me is-- 
Papua New Guinea FTO Chuave - Keto Tapasi - AX_ 17C_5.0oz_20-05-02_1319_R23 (2).alog

While I haven't tried all the possibilities, it has been working great for me. 

Special Events Annotations
I've only used the '100% Event Step' feature so far; this works nice. I'll try some of the other capabilities later. 

I'll keep trying things and provide feedback. I haven't trying doing an actual roast yet. 

- Dennie

MAKOMO, 2020-05-19:
Dennie, thanks for this detailed report. Some clarifying questions and comments:

- alignment on TP is technical not possible and also does not make much sense to me as the TP is just an artefact of the measuring system or rather its error. I would say: irrelevant
- Comparator: regarding that double click thing: are you working on Windows or macOS?
- Simulator:
  - the autosave is standard, right? what is wrong with it?
  - that standard sliders are recorded is also expected. What is wrong with it?
  - the delta lines are only drawn after the CHARGE event has been set. This is the same behavior as in regular roasts
  - events/slider events not displayed: maybe there is some confusion. The simulator is not replaying events. It is just taking the temperature/curve data from the given profile instead of a connected device. All events have been entered as before manually. Thus Artisan behaves exactly as with a real roast just that the data comes from a different source and also no other communication happens with connected devices

All other things seem to work for you, as far as you have tested them, right?

Thanks for this review. Dave will take a second look into this inter-app communication that seems to fail w.r.t. this Comparator row double click thing.

DennieSch, 2020-05-19:
- TP; understand, we had an earlier discussion on TP. My thinking was (and that is what I was testing with the comparator) was comparing several roasts of the same beans but with different charge weights. So, my thinking was the better point to align on in those cases is TP. Otherwise, the roast graphic are skewed due to charge weights.

- Comparator; Architecture = ('64bit', 'WindowsPE')
Machine = AMD64
Platform name = Windows-10-10.0.18362-SP0

- Simulator; my assumption was the Simulator is intended replay the roast in its entirety (including all data and events). I think I'm not understanding what the intended use is for the Simulator or... how to use it properly. For example, why would I record a different Charge time, Dry time or FCs than what was originally recorded. That isn't going to change the actual BT/ET and delta data is it. 
The description does say "...alarm rules and the replay of events". So, I was expecting events to be replayed. I don't recall seeing my alarm pop up but this could be related to #462 

Otherwise, things have been working thus far.

MAKOMO, 2020-05-19:
Simulator: as said, Artisan just takes (temperature) data from that given profile and behaves (or should behave) otherwise as if it get this data from a connected device. The purpose of the simulator is to test new features, eg. alarm rules, symbolic formulas and such, not to repeat a profile (why should you?). For this testing, data is needed, but you might not want to waste beans for this. Events are replayed from a background profile if you activate that feature and have a background profile loaded, as with any normal roast. Same holds for alarms. If your alarms did not trigger that might be related to the lack of a CHARGE event being set. Issue #462 is fixed.

DennieSch, 2020-05-19:
Thanks Marko, that gives me a much better understanding of the intent. I haven't used background profiles other than just loading one to see what it looks like. Knowing all of this I'll give it another go. Thanks!

DennieSch, 2020-05-19:
Marko, I don't know if this means anything but thought I would point it out 
The Release Page says the 2.4.0 RC commit is  eb1cb8c however, the About in the application shows this; version = 2.1.3 (532987c5) a different commit serial number

MAKOMO, 2020-05-20:
Yes, I was a bit lazy on that. I reused a previous release that I made for an initial RPi buster build some days ago and I also didn't push the version number for this prerelease to minimize troubles on my side. Shouldn't harm.

DennieSch, 2020-05-22:
Unable to 'OK' or accept any changes made to any of the tabs of the Config|Curves... menu due to issue #459. No Minimize or Maximize buttons on window title bar either. 
(I wanted to make changes to my Filter settings but unable make the change since I could access the 'OK' button.)