artisan-roaster-scope/artisan #468: Call to Program Python does not seem to work

reynwouda, 2020-05-20:
## Expected Behavior
Call the python test program and logs value of slider to a log file

`
#! C:\Python\python.exe
import sys

allArgs = sys.argv

test = open("logs.txt", "w")
test.writelines(str(allArgs))
test.close`

## Actual Behavior

Have a simple python program to test if i can grab the values from my slider 
when i adjust the slider values but the event does not seem to call the program at all 

## Steps to Reproduce the Problem

  1. added a slider for air 
  2. select option call program under action 
  3. under commands i have the following
      D:\development\devlab\lab_program\start.py 18 {} testargs
 4. then is start the roast and adjust the slider 
 5. when i look at log file it is empty 

i did test this outside the program and the script seems to run just fine 

## Specifications

  - Artisan Version:2.1.2
  - Artisan Build (number in brackets shown in the about box): 2d7c9951
  - Platform (Mac/Windows/Linux + OS version): Windows-10-10.0.18362-SP0
  - Connected devices or roasting machine: Self Build roaster 
[artisan-settings.aset.txt](https://github.com/artisan-roaster-scope/artisan/files/4657644/artisan-settings.aset.txt)
`

  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

roundfile, 2020-05-21:
The system is looking for your Python executable to be at "C:\Python\python.exe" and not finding it.  You just need need to put the correct full path to python.exe in front of your script path.  For example:
`"c:\python38-64\python.exe" "D:/development/mervyn_roaster/lab_program/dummyTemp.py"`

If the Python executable is in your environment PATH then you can simply put "python.exe" in front like this:
`"python.exe" "D:/development/mervyn_roaster/lab_program/dummyTemp.py"`

reynwouda, 2020-05-21:
Hi

Thanks for the response my temperature program works perfect i did not have
issues with the dummyTemp.py running under CONFIG --> Devices --> Progs

What i have a issue with is getting the same result calling a python script
from a Event --> Slider

This small program will just log the values i enter in the event command
section in a log file

in the event command section i have the below command

*D:\development\mervyn_roaster\lab_program\start.py 12 {} 35*

My path to my interpreter is correct c:\Python\python.exe is my full path

my python code for slider is as follows
#! "c:/Python/python.exe"
*import sys*

*allArgs = sys.argv*

*test = open("logs.txt", "w")*
*test.write(str(allArgs))*
*test.close*

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#m_9095656944139476831_m_-7781069841855246897_DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

-- 

Kind Regards

Reyn Wouda
Senior Internet Engineer

Skype: reynhard.packetsky
Web:    PacketSky.com

This e-mail is confidential and intended solely for the use of the
individual to whom it is addressed. Any views or opinions presented are
solely those of the author and do not necessarily represent those of
PacketSky. If you are not the intended recipient, be advised that you have
received this e-mail in error and that any use, dissemination, forwarding,
printing, or copying of this e-mail is strictly prohibited. If you have
received this e-mail in error please notify PacketSky by telephone, fax or
e-mail. PacketSky is not held responsible for loss or damage due to content
of this e-mail.

reynwouda, 2020-05-21:
Hi

Thanks for the response my temperature program works perfect i did not have
issues with the dummyTemp.py running under CONFIG --> Devices --> Progs

What i have a issue with is getting the same result calling a python script
from a Event --> Slider

This small program will just log the values i enter in the event command
section in a log file

in the event command section i have the below command

*D:\development\mervyn_roaster\lab_program\start.py 12 {} 35*

My path to my interpreter is correct c:\Python\python.exe is my full path

my python code for slider is as follows
#! "c:/Python/python.exe"
*import sys*

*allArgs = sys.argv*

*test = open("logs.txt", "w")*
*test.write(str(allArgs))*
*test.close*

reynwouda, 2020-05-21:
Lol seems like i was just being dumb i added the full path of the text file to the python script and now it works perfectly

closed 2020-05-21 by reynwouda