artisan-roaster-scope/artisan #475: New version 2.4 for Mac OSX does not work with the Hottop KN-8828B-2K+

mfarre, 2020-06-13:
## Expected Behavior
Hottop KN-8828B-2K+ working as it was in version 2.1.2

## Steps to Reproduce the Problem

  1. Download the new release of artisan
  2. Select the Hottop KN-8828B-2k+ as device
  3. The temperature values are -1

## Specifications

  - Artisan Version: 2.4
  - Artisan Build (number in brackets shown in the about box): 5afdae1
  - Platform (Mac/Windows/Linux + OS version): Mac OS 10.15.5
  - Connected devices or roasting machine: Hottop KN-8828B-2k+
  

I have also downloaded the source code of the latest version and ran it directly and that works with no issue but for some reason the packaged version does not work.

Here the library versions of the packaged version of artisan (this version is the one that does not work with the hottop)

Artisan 2.4.0 (5afdae1) 
Python 3.8.3, Qt 5.14.2, PyQt 5.14.2, Matplotlib 3.2.1, NumPy 1.18.4, SciPy 1.4.1, pymodbus 2.3.0, Yoctopuce 1.10.40411 (1.10.40411) 

Here the library versions when I run the code in the git repo directly in my computer and actually works:
ArtisanViewer 2.4.1 (0) 
Python 3.7.7, Qt 5.15.0, PyQt 5.15.0, Matplotlib 3.1.1, NumPy 1.18.2, SciPy 1.4.1, pymodbus 2.3.0, Yoctopuce 1.10.40411 (1.10.40411) 

Maybe the python version in the packaged version of artisan is the problem?

mfarre, 2020-06-13:
Update: using the requirements.txt and requirements-os.txt I got locally running the same versions of all the libraries than the ones of the packaged release:

Artisan 2.4.1 (0) 
Python 3.8.3, Qt 5.14.2, PyQt 5.14.2, Matplotlib 3.2.1, NumPy 1.18.4, SciPy 1.4.1, pymodbus 2.3.0, Yoctopuce 1.10.40411 (1.10.40411)

but running it directly `python3.8 artisan.py` the hottop works and not with the packaged version of the app.

as a side note, I see this warning in the terminal:
/usr/local/Cellar/python@3.8/3.8.3/Frameworks/Python.framework/Versions/3.8/lib/python3.8/multiprocessing/resource_tracker.py:216: UserWarning: resource_tracker: There appear to be 1 leaked semaphore objects to clean up at shutdown
  warnings.warn('resource_tracker: There appear to be %d '

MAKOMO, 2020-06-14:
Confirmed. This one is related to a change in multiprocessing of Python3.8 vs Python3.7 on macOS.

MAKOMO, 2020-06-14:
The new continuous builds do contain the fix: https://github.com/artisan-roaster-scope/artisan/releases

closed 2020-06-14 by MAKOMO