artisan-roaster-scope/artisan #477: artisan.exe Flagged by Malwarebytes

LiddleNut, 2020-06-15:
Hi, I installed Artisan 2.4.0 11 days ago.  All daily scans previous to today had no issues.  Today Malwarebytes (4.1.0 premium) quarantined artisan.exe and identified the issue as [https://blog.malwarebytes.com/detections/spyware-passwordstealer/](url)

TIA

MAKOMO, 2020-06-15:
Artisan spans a background process for WebLCDs or the Hottop communication. It might happened that this process got hanging on your machine beyond the termination of Artisan. This hanging process might have been categorized as a passwordstealer in the sense of your tool. You can check the Artisan code for any paswordstealing code as it is contained in this repository. The builds are made on a clean VM as always and cannot lead to infected builds. For sure a false alert.

LiddleNut, 2020-06-15:
I posted the following on the MB forum.  TY for your help.
[https://forums.malwarebytes.com/topic/260767-spywarepasswordstealer-artisan/](url)

LiddleNut, 2020-06-16:
MB Staff response:

> Thanks for reporting, this has already been fixed.

Best Regards

closed 2020-06-16 by LiddleNut