artisan-roaster-scope/artisan #482: I can't run version 2.40

CarlosAlbertoRamos, 2020-06-19:
## Steps to Reproduce the Problem

  1. install version
  2. when i try to run nothing happens not even error
  3. 

## Specifications

  - Artisan Version: 2.4.0
  - Artisan Build: I can not see
  - Platform: 
  - Connected devices or roasting machine:
  Windows 10 home version 1909 compilation 18363.900 with processor Intel i3-7100U
  Laptop Dell Inspiron 5566

Note: Version 2.1.2 runs normally

roasterdave, 2020-06-19:
Please try the following.  Assuming you install to the default location, rename the folder c:\Program Files\Artisan to something else.  Then again install v2.4.0.  Click OK if you get the message that Artisan is already installed.  Please report back on the results.

CarlosAlbertoRamos, 2020-06-20:
Nothing changed. The weirdest thing is that it runs on my daughter’s laptop, which has the same operating system, only changes the processor. Maybe the processor is the problem? Or the Dell laptop? I don't know I'm discouraged and I've heard another report of the same problem I just don't know if it's same model Dell too.

roasterdave, 2020-06-20:
We have checked the libraries and Python versions used and while there are differences we cannot find changes to explain this.  Perhaps there is some connection with the processor but I cannot find it.  The closest I can get to your system is a Dell i5 where the install works fine.  If you are willing to help me dig a little to resolve this please contact me via email at issue482@gmx.com.  Thanks, Dave

murilosantos09, 2020-07-06:
Carlos, were you able to make Artisan work? I am having exactly the same problem with the 2.4 version.

roasterdave, 2020-07-06:
Murilo, please try the continuous build available here https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous and let me know if this resolves the issue for you.
Dave

CarlosAlbertoRamos, 2020-07-06:
Hi Murilo,
I only managed to run version  "continuos buildind" that you can find on the link All Releases
Despite the warning that this type of version may not work it is working perfectly. I think it was a problem on dell laptops with i3 2.40 Grz processor. Anyway for me resolved so I suggest you download the version.

CarlosAlbertoRamos, 2020-07-06:
> Murilo, please try the continuous build available here https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous and let me know if this resolves the issue for you.
> Dave

Yes, for me it was the solution. sorry I didn't post here

closed 2020-09-25 by MAKOMO