artisan-roaster-scope/artisan #483: Can't connect to Hottop with 2.4 on Mac

drewconner, 2020-06-23:
When I select my device (Hottop KN-8828B-2K+), Artisan defaults to a port of /dev/cu.usbserial-DA01POGR. My Hottop connects to /dev/cu.usbserial-DN017CMR. With all previous version of Artisan, I've been able to simply edit the port name and all has worked well. Starting with 2.4, the edit no longer seems to apply and I can no longer connect.

A downgrade to 2.1.2 allowed me to edit the port again and roast successfully.

MAKOMO, 2020-06-23:
On which platform are you. At least on macOS I can edit the port easily. The correct port should also be in the popup while your roaster is connected. Could you give us a little more information on this?

MAKOMO, 2020-06-23:
Note that there is a regression in v2.4 that breaks the communication with the Hottop (indepdent of the serial port setup). See #475. The current continous builds should work fine for now until we release a 2.4.2 bug fix release https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous.

thepilotusa, 2020-06-28:
Anyone who might have problem connecting your Mac to Hottop, check if you have Artisan version 2.4.0. If you do, uninstall and install version 2.1.2. For me it worked immediately!

drewconner, 2020-06-30:
I just tried the latest continuous build and it does indeed work again. Thanks!

closed 2020-06-30 by drewconner

philipp1992, 2020-07-08:
had the same problem, maybe its better to keep this issue open until its fixed