artisan-roaster-scope/artisan #484: Alarm not working as expected - Windows 10 - Artisan version 2.1.2 - Hottop B KN-8828B-2K+

kishoregd, 2020-06-24:
## Expected Behavior
Alarm was set after TP to execute a Hottop Command
Alarm Setup:

If Alarm = 0
But Not = 0
From = TP
Source = BT
Time = 00.00
Condition = above
Value = 463.0
Action = Event Button
I created a new button in the events and trying to click the button via Alarm

## Actual Behavior
This Alarm is getting fired between 436 and 438 BT values.

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version: 2.1.2
  - Artisan Build (number in brackets shown in the about box): 2d7c9951
  - Platform (Mac/Windows/Linux + OS version): Windows 10
  - Connected devices or roasting machine: Hottop B KN-8828B-2K+
[artisan-settingsaset.txt](https://github.com/artisan-roaster-scope/artisan/files/4823145/artisan-settingsaset.txt)
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2020-06-24:
Your alarm and button definitions look ok. You did not attach any profile that demonstrates that the alarm was triggered at a too low temperature, which makes it impossible to investigate. There are many other alarms that look quite similar but have different actions. Those trigger all correctly!?

MAKOMO, 2020-06-24:
Note that you are using an outdated version of Artisan!

MAKOMO, 2020-09-16:
Stall

closed 2020-09-16 by MAKOMO