artisan-roaster-scope/artisan #485: Delta BT always -1.0 with version 2.4 on Hottop KN-8828B-2K+

ChococarSB, 2020-06-26:
Delta BT should be calculated as the slope of BT over time.  BT is calculating and displaying correctly

Delta BT is always displayed on LCD as -1.0

Problem occurs every time.  (Delta ET is calculated and displayed ok)

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version: 2.4.0
  - Artisan Build (number in brackets shown in the about box): 164de804
  - Platform (Mac/Windows/Linux + OS version): Windows 10
  - Connected devices or roasting machine: Hottop JB-8828B-2K+
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.
[artisan-settings 062620.aset.txt](https://github.com/artisan-roaster-scope/artisan/files/4838397/artisan-settings.062620.aset.txt)

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2020-06-26:
Remove the wrong formula "ROR_BT" in the field "\DeltaBT Y(x)" under menu Config >> Curves, 1st tab and your Delta BT is again displayed correctly.

closed 2020-06-26 by MAKOMO

ChococarSB, 2020-06-26:
Yay!  Thank you very much!!

Thanks,
Joe