artisan-roaster-scope/artisan #486: ET/BT External Program freezes Artisan

normaldotcom, 2020-06-28:
When specifying a Python script as the external program for ET/BT, Artisan appears to successfully launch a Python process but then hangs The "Off" button no longer works and things lock up after a few minutes.

The python script is simply a CSV print statement loop with a 1 second delay for testing. This is also repeatable when I remove the print statement and the Python script simply loops on time.sleep(1).

I have reproduced this on both Linux and Windows with Artisan 2.4.0.

MAKOMO, 2020-06-29:
You did not share your script here. Did you specify the full path to the Python app in your script? Note that the script is evaluated in an empty environment for reasons.

normaldotcom, 2020-06-29:
My script:

```
import time

while True:
    time.sleep(1)
```

[image]

My external program path is C:/temp/test2.py
I have also attempted setting the path to C:\Windows\py.exe C:\temp\test2.py and the same issue occurs.

I also reproduced this on another Windows computer, also with Artisan 2.4.0.

Artisan successfully launches a Python process but then freezes.
[image]

roasterdave, 2020-06-29:
Hi, 

The external program is intended to print data that is read by Artisan and then exit.  It should not be persistent.  Artisan starts the external program once each sample period.  Try a simple one line script that returns a static ET and BT value.  This will demonstrate the ability to start the program, transfer the data and plot it each sample period.  For example, 
`print ("237.1,200.4")`

The path you used should work fine.  
[image]

[image]

normaldotcom, 2020-06-30:
That makes complete sense! Do you think it would be worth me submitting a PR to update the "help" button text with a sentence that says the program must print to stdout and exit, not be persistent?

closed 2020-06-30 by normaldotcom

roasterdave, 2020-06-30:
Thanks for the PR offer!  In this case the change is simple enough, I've already modified the help file to include in a next commit.