artisan-roaster-scope/artisan #487: Enhancement: Use alternative to system beep for alarms

rlabbe, 2020-06-29:
I want this for two reasons. One, I have the default sound theme on my computer set to nearly silent. I don't want to be beeped at me continuously, and so default beep is set to "none". Second, a single beep is not very noticeable. I have my alarms set up to alert me to get ready to load beans, for example. I want a very persistent alarm so I can't miss it. I set it to the obnoxious alarm01.wav sound on my pc, which gets my attention for the 3 seconds it plays, but now every time I click in the wrong place or whatever I get this same sound. Ugh.

The source code currently uses QApplication.beep() if you ask for a beep with your alarm. This triggers the system default sound It would be great if we had an option to select a wav file for the alarm. You should be able to use QSound.play(filename) to play the wav. Even better would be to offer the option to have it loop; the alarm dialog that pops up would either call QSound.stop() when closed, or offer a button to silence the alarm. 

This may not be entire cross platform compatible, but I would imagine nearly any system running Artisan can play wav; if not, the default beep will still work (in my scheme, I'd use the default beep if the user did not specify a custom alarm sound). 

I poked around, it looks like you aren't currently packaging QtMultimedia, so that would be an additional import, which might be a pain if you are using conda (for whatever reason conda doesn't include QtMultimedia).

das3in, 2020-09-09:
This is a cool idea. I think there's a lot of hidden complexity in a file upload system, so a decent first pass might be to:

* Customize beep with a `.wav` file
* Ship Artisan with, say, 3 different beeps
* Expose the folder location on the wiki so people can drag and drop their own files we don't need to worry about upload/validation infrastructure.

I'll poke around at this this weekend. Good idea!

MAKOMO, 2020-09-09:
Not sure if Artisan needs to be modified for this. macOS offers the shell command `afplay` that can play any type of sound files and you can configure an alarm to call that command like many are currently using the command `say` to make alarms talk. I am quite sure that something like this exists on the Windows platform as well.

reizao, 2020-10-08:
Yes. In windows you can call app espeak