artisan-roaster-scope/artisan #488: None of the new version of Artisan work with my Phidget 1048_0 4 input device.

evermorecoffee, 2020-07-01:
## Expected Behavior
Will be able to see accurate temperature readings when connecting to the device, as I have been using versions 1.4v of artisan and had no issues. I would just like to be able to utilize the newer User interface and try out the inventory system, but I can't seem to figure out how to configure the device. 

## Actual Behavior
I see no temperature data, and negative temperature readings. I have a brand new version of the software installed

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version:
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version):
  - Connected devices or roasting machine:
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2020-07-01:
Did you install the current Phidgets driver v22 (note that some systems like Cropster still relay on the very outdated v21 driver which is incompatible with all newer Artisan versions).

closed 2020-07-01 by MAKOMO

evermorecoffee, 2020-07-01:
Hey Marko,

Thank you for that.

Best regards,
Ryan