artisan-roaster-scope/artisan #490: MODBUS Register must be addition with 5 from lowest address number?

hidasuryo, 2020-07-10:
## Expected Behavior

I am develop Modbus device planed with 7 sensors (ET, BT, Furnaces/Burner Temp, Gas debit flow, Gas Valve Position, Drum RPM and Air Flow), and for now already run with 5 sensor.  I am glad, the artisan 2.4.0 ready for 8 sensor. But i am find the problem, register numbering totally different with previous version (2.1.2 and before's), that we can determine register number in order like 1,2,3,4,5,6. 
In 2.4.0 version why must be adding by 5 in order like 1, 6, 11, 16, 21 ?  or like 2, 7, 12, 17, 22 ?
## Actual Behavior

is it possible to return the register numbering as before ? because many have used mu device in Indonesia.

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version: 2.4.0
  - Artisan Build (number in brackets shown in the about box): 164de804
  - Platform (Mac/Windows/Linux + OS version): Windows 10
  - Connected devices or roasting machine: ATS-02 (my won develop, with Modbus TCP Connection
)
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.

[ATS02_24 .txt](https://github.com/artisan-roaster-scope/artisan/files/4903601/ATS02_24.txt)

Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2020-07-10:
The MODBUS register numbering scheme did not change between versions. We only added two more channels and the corresponding configuration. Not sure what exactly your issue is. Sorry.

zahidintegra, 2020-07-10:
from your file setting configuration :

[Modbus]
comport=COM6
baudrate=115200
bytesize=8
stopbits=2
parity=N
timeout=1
PID_slave_ID=0
PID_SV_register=5
PID_p_register=0
PID_i_register=1
PID_d_register=0
PID_OFF_action=
PID_ON_action=
input1slave=1
input1register=2
input1float=false
input1bcd=false
input1code=4
input1div=2
input1mode=C
input2slave=1
input2register=7

Do you set your device register correctly?

 it s*eems depend on your device register mapping*

*rgds *

MAKOMO, 2020-07-10:
Well this is from the configuration file you attached, not me. These are just the application settings that store the registers as specified in the MODBUS tab. You specified register 2 for Input channel 1 and register 7 for input channel 2 and so on. If you need to access different registers, configure different registers. I still don't see the problem. Sorry.

[image]

What changed between the previous versions and Artisan v2.4 is that now data is requested in blocks. So if you use data from register 1, 2 and 3 Artisan will send a request to get data from those 3 registers in one requests starting from register 1. If you use data from registers 1, 3 and 4. Two requests are issued. One to request the data from register 1 and another one to get data for 2 register with start register 3. Earlier Artisan versions were sending one request per input/register. The new approach of requesting data in contiguous registers together reduces the amount of requests to be send and thus speeds up the communication. This is pretty standard and specified for all MODBUS functions. Eg. see the specification of MODBUS function here: http://www.simplymodbus.ca/FC03.htm. You specify a start register and the number of registers to read from.

hidasuryo, 2020-07-10:
Thanks for All response...
here is the screen capture serial monitor from my device...
if i use 2,3,4,5,6 register, only register 2 requested by artisan received my device.
but if i am use 2, 7, 12, 17, 22 register, all registers requested by artisan received, and my device.
[image]
[image]

hidasuryo, 2020-07-10:
> Well this is from the configuration file you attached, not me. These are just the application settings that store the registers as specified in the MODBUS tab. You specified register 2 for Input [...]

noted...
I try connection with Serial RTU and register use 2,3,4,5,6 all work fine.
so i check the library for Serial RTU and TCP that I use. in Serial RTU library, it ready to handle more than 1 register's request, but in TCP library, I have to adjust my code to handling more than 1 register request.

ok.. i understand now, Artisan v2.4 will request sequential registers at once. its good for speeds up the communication.

Thanks for your response..

closed 2020-08-30 by MAKOMO