artisan-roaster-scope/artisan #492: v2.4.0 freezes during config and when trying to launch new roast

roasterracer, 2020-07-12:
Have new laptop - trying to configure artisan to roaster via bluetooth (Arc 800) and the program consistently enters a "Not Responding" state. Following instructions provided by Arc/Crop to Cup, whenever trying to select Port from menu, Artisan freezes. Artisan will also freeze after setting Device (to TC4) and adjusting settings - which then pushes to Port. Have encountered issue both when running program normally, and then also when as Administrator.

Artisan also doesn't click into drop downs, and does not turn "off" after turning "On" to read ET/BT probes of Com port. However, this is less important atm as I can't even configure the roaster to the program. Have been using Artisan and use old laptop (2016 Macbook Pro) in order to log roasts. Will try continues build to see if I can get configured and run roasts.

Setup:
Asus FA506, Win 10 64bit, AMD Ryzen 7 4800H 2.9ghz, 32GB RAM

## Specifications

  - Artisan Version: 2.4.0
  - Artisan Build (number in brackets shown in the about box): build 1
  - Platform (Mac/Windows/Linux + OS version): Win
  - Connected devices or roasting machine: Arc 800
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2020-07-12:
Could you please attach your Artisan settings file (save via menu Help >> Save Settings) and maybe attach those instructions you received from Arc/Crop to Cup to check.

Does Artisan works on a USB connection to the roaster?

Does a Artisan v2.1.2 works for you on your Windows machine? Does Artisan v2.4.0 works for you on your MacBook?

MAKOMO, 2020-07-12:
Did you install any USB driver on your Windows PC to connect to that ARC roaster?

roasterracer, 2020-07-13:
Hi Makomo

Sure. I've attached the settings from my MBP and the Asus/Windows laptop that's giving me the issues. I've also attached the instructions I've received from Crop to Cup. MBP is on v2.0.0 (a0a1fac) and not on 2.4. 

I haven't tried connecting via USB to the roaster since Bluetooth worked on the MBP. I did some scrounging and found a USB to USB C that I can try later tonight. I have not installed any USB drivers on the Asus but I will install the following Windows 10 driver per the instructions in the Arc roaster manual by Silicon Labs which can be downloaded here https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers. 

Have not tried v2.1.2 on the Asus/Windows machine but will install and also try that later tonight. 

Appreciate your help

[artisan-settings-macbook-pro.txt](https://github.com/artisan-roaster-scope/artisan/files/4914503/artisan-settings-macbook-pro.txt)
[Arc Roaster User Manual - Gas 011718.pdf](https://github.com/artisan-roaster-scope/artisan/files/4914505/Arc.Roaster.User.Manual.-.Gas.011718.pdf)
[artisan-settings-asus.txt](https://github.com/artisan-roaster-scope/artisan/files/4914508/artisan-settings-asus.txt)

roasterracer, 2020-07-14:
Installed v2.1.2 and attempted again. Still freezes after selecting Port. My USB to USB C cable didn't recognize the roaster so I'll have to pick up a new cable to try the USB connection.

MAKOMO, 2020-09-05:
Could you resolve this one? Note that the continuous builds now contain an extended setup for the Arc 800 machine that reports 4 instead of just 2 temperatures. Could you try that one and report back?

roasterracer, 2020-09-05:
I ended up returning the laptop. I couldn't risk keeping a laptop that
wouldn't work with Artisan for my roasts. I'm still just using the MBP in
the interim until I have time to look for another laptop

MAKOMO, 2020-09-05:
Ok. So it is rather an  issue with that hardware, maybe not enough memory. Could you test those new builds on your Mac and see if those work better with your Arc 800. All you have to 

1) save your Artisan settings just in case something goes north (menu Help >> Save Settings)
2) download the beta build from here: https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous
3) start that version of Artisan and configure for the Arc 800 while connected to your machine (menu Config >> Machine >> Arc 800)
4) press ON and see if you get all 5 temperatures displayed correctly

Please report also if all goes well. If things do not work out, you can just start your previous Artisan version (or download/install again the v2.4.0) and load the settings file via menu Help >> Load Settings that you saved in step 1) to get back to your original state.

Thanks for testing and your support,
Marko

roasterracer, 2020-09-07:
Hi Marko

Thanks for this build. All 5 data points appeared on the right hand side of
the screen. BT, ET, Drum speed, Air Inlet temp (i assume) and AT (not sure
what this one is, but it was consistent w/ drum speed. pardon the messy
roast profile. quick question, is there a way to set which graph line
should appear on top? I haven't looked through the documentation in a while
and Drum speed and Delta ET appear over Delta BT, which i believe is the
most important and should always be on top.

best

tom

[image: Screen Shot 2020-09-07 at 2.52.31 PM.png]

roasterracer, 2020-09-07:
my mistake it's not Drum speed, i thought thats what it was because I keep
my drum speed the same usually during the roast. I'm asking the Crop2Cup
team what the Drum & AT sensors do and what they read, as well as where the
Inlet temp sensor is and what it's actually measuring.

MAKOMO, 2020-09-07:
Thanks for testing!
Note that the screenshot did not get attached. Could you  attach the corresponding .alog file (rename to .txt) for me to analyze the data?

AT should be Ambient Temperature. Not sure where it is measured. What "Drum" measures remains unclear to me.

Temperature curves are always rendered above RoR curves. By default ET is rendered above BT (can be swapped via Config >> Device, 1st tab) and DeltaET above DeltaBT (can be swapped via Config >> Curves).

roasterracer, 2020-09-07:
ahh i replied back via gmail. i attached the .alog as .txt if it doesn't
come through i'll repost manually through the forum

{'version': '2.4.1', 'revision': 'd6d5229', 'build': '0', 'artisan_os': 'Mac OS X', 'artisan_os_version': '10.14.6', 'mode': 'F', 'timeindex': [14, 124, 233, 268, 0, 0, 312, 0], 'flavors': [5.0, 5.0, 5.0, 5.0, 5.0, 5.0, 5.0, 5.0, 5.0], 'flavorlabels': ['Acidity', 'Aftertaste', 'Clean Cup', 'Head', 'Fragrance', 'Sweetness', 'Aroma', 'Balance', 'Body'], 'flavorstartangle': 90, 'flavoraspect': 1.0, 'title': 'Brazil', 'beans': '', 'weight': [610.0, 0.0, 'g'], 'volume': [0.0, 0.0, 'ml'], 'density': [0.0, 'g', 1, 'l'], 'density_roasted': [0.0, 'g', 1, 'l'], 'roastertype': 'Arc 800', 'machinesetup': '800', 'operator': 'T Morsanutto', 'organization': '', 'drumspeed': '55', 'heavyFC': False, 'lowFC': False, 'lightCut': False, 'darkCut': False, 'drops': False, 'oily': False, 'uneven': False, 'tipping': False, 'scorching': False, 'divots': False, 'whole_color': 0, 'ground_color': 0, 'color_system': '', 'volumeCalcWeightIn': '', 'volumeCalcWeightOut': '', 'roastdate': 'Mon Sep 7 2020', 'roastisodate': '2020-09-07', 'roasttime': '14:41:10', 'roastepoch': 1599504070, 'roasttzoffset': 18000, 'roastbatchnr': 0, 'roastbatchprefix': '', 'roastbatchpos': 1, 'roastUUID': '8017d2383f914106acb5c74fe9ced11b', 'beansize': '0.0', 'beansize_min': '0', 'beansize_max': '0', 'specialevents': [], 'specialeventstype': [], 'specialeventsvalue': [], 'specialeventsStrings': [], 'etypes': ['Air', 'Drum', 'Damper', 'Burner', '--'], 'roastingnotes': '', 'cuppingnotes': '', 'timex': [1.39573627, 3.402823247, 5.392951775, 7.3844380435, 9.4031658105, 11.392058277, 13.385555179, 15.4024734165, 17.3823696495, 19.382493499, 21.4052255605, 23.39290219, 25.3803431725, 27.404323775, 29.3928751255, 31.382751311, 33.4032668, 35.4227865515, 37.3820875275, 39.405852792, 41.392687669, 43.381988675, 45.40259657, 47.438286224, 49.39024238, 51.403273402, 53.3955452345, 55.3836537335, 57.4036159575, 59.3927269035, 61.3838039135, 63.402239513, 65.401111244, 67.3828341615, 69.40154643, 71.392715449, 73.3815450105, 75.4046128405, 77.3914456485, 79.382042212, 81.4032185745, 83.3932168275, 85.3831927975, 87.400813713, 89.376318457, 91.3824990275, 93.401235276, 95.391163825, 97.390542928, 99.400645184, 101.3925091945, 103.3808515065, 105.4047360045, 107.392270147, 109.3817747865, 111.400941285, 113.3985710775, 115.38392589, 117.400375606, 119.394263087, 121.381620033, 123.4004274675, 125.3908297255, 127.380687747, 129.4016309665, 131.391255498, 133.3836682525, 135.4025110635, 137.392460171, 139.3804645305, 141.400481387, 143.3904011405, 145.391122157, 147.4004580595, 149.3901371935, 151.379624191, 153.4008777845, 155.39276014, 157.3827307915, 159.405119668, 161.397686724, 163.3813201945, 165.4004209835, 167.384347082, 169.3799775695, 171.3997438865, 173.3899127475, 175.379646059, 177.4004423635, 179.389782745, 181.380407986, 183.3999438995, 185.391691983, 187.3802158885, 189.3696369965, 191.390223728, 193.3920495085, 195.400334946, 197.389516706, 199.3833230075, 201.399208644, 203.391348083, 205.381224498, 207.4000000725, 209.396822693, 211.380569592, 213.400545154, 215.390454105, 217.3791696285, 219.398937278, 221.3891796975, 223.3791341745, 225.3997185435, 227.390839433, 229.3791574845, 231.399606887, 233.388687157, 235.38082208, 237.399505108, 239.383821286, 241.3769680385, 243.4304741715, 245.3893290775, 247.379000634, 249.4019249635, 251.390021223, 253.378756129, 255.3992355145, 257.3743753955, 259.380426048, 261.4004293285, 263.3903797905, 265.4103437135, 267.4002539735, 269.389403853, 271.418351349, 273.400174411, 275.420829057, 277.37828658, 279.3982760705, 281.388916242, 283.3786190915, 285.3981540365, 287.3906020315, 289.413708329, 291.429231925, 293.3899607185, 295.4078969315, 297.399859464, 299.3873454635, 301.410458139, 303.3990802395, 305.3960618895, 307.3785416765, 309.397847307, 311.3878800585, 313.407788492, 315.398312696, 317.384047349, 319.3789045165, 321.398825657, 323.418109148, 325.4082093225, 327.397569492, 329.387500367, 331.379898411, 333.3973548995, 335.387304995, 337.389109982, 339.4268375475, 341.3876245505, 343.4084700305, 345.397251101, 347.387707214, 349.377804354, 351.3978160395, 353.394615036, 355.37903429, 357.396987658, 359.3895062595, 361.407536587, 363.396853907, 365.3875725735, 367.4162714105, 369.3974226805, 371.4167720015, 373.406252552, 375.399190597, 377.4173520655, 379.4067090445, 381.3966493575, 383.3872601925, 385.4162857295, 387.426587283, 389.3884631375, 391.4077806115, 393.3977218715, 395.3870925075, 397.406805572, 399.396415353, 401.396321808, 403.3762617405, 405.400000193, 407.3881571405, 409.406116611, 411.3999681205, 413.3717726525, 415.414279465, 417.3962669965, 419.418587966, 421.3760584255, 423.396658279, 425.386606176, 427.4059796835, 429.3969530815, 431.3883106665, 433.375262895, 435.3982501975, 437.4089584265, 439.412616248, 441.39946152, 443.3863472135, 445.4069073825, 447.3981248035, 449.3937272155, 451.405482883, 453.3961240975, 455.3867607995, 457.4054045805, 459.39611893, 461.3872846915, 463.416037631, 465.401617844, 467.415266742, 471.3952645715, 473.3853135725, 475.37513597, 477.3963911585, 479.385256516, 481.43644991, 483.3964217615, 485.3868184685, 487.4011510625, 489.3956272135, 491.384835059, 493.404902583, 495.394870031, 497.394102355, 499.4059754245, 501.3965562055, 503.3872723555, 505.4080677575, 507.3953848555, 509.384572691, 511.4132530675, 513.395859366, 515.4 [... cut ...] mbient_temperature': 107.1, 'det': 15.044611353018695, 'dbt': 18.89276672275547}, 'anno_positions': [[0, 29.392875125499813, 488.37, 29.392875125499813, 388.37], [-1, 117.4003756059999, 289.03999999999996, 117.4003756059999, 189.04], [1, 249.40192496349994, 355.28, 249.40192496349994, 255.27999999999997], [2, 467.41526674199986, 448.5, 467.41526674199986, 348.5], [3, 539.3841588445, 479.47, 539.3841588445, 318.47], [6, 627.4238612475001, 509.72, 627.4238612475001, 387.72]], 'flag_positions': []}

MAKOMO, 2020-09-08:
Got the alog file. I wonder about those horrible temperature signals which show quite some steps. This is not due to the lack of decimals, but because some readings are just repeated. Could this be caused by a communication issue? Could you please support me to dig down this and open the serial log (menu Help >> Serial Log) and tick the flag there and then record some seconds where the temperature is rising, just to see what raw data is received from the machine? Please cut-copy the result from the serial log window here. Thanks!

MAKOMO, 2020-09-09:
Hm. Checked the data and it seems the machine is sending every second or third request same old data as previously send. That creates data with wrong timestamps in Artisan which is rendered as those steps.

[image]

I added an input filter to take out those duplicate readings and interpolate them with the following readings. Now the same roast looks like this.

[image]

Would you mind testing this with your machine. You need to download the latest build from here 

https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous

and again choose menu Config >> Machine >> Arc >> Arc800. Please repot.

roasterracer, 2020-09-13:
still kinda steppy during the roast, but post roast analysis it's
smoothed out

{'version': '2.4.1', 'revision': 'd6d5229', 'build': '0', 'artisan_os': 'Mac OS X', 'artisan_os_version': '10.14.6', 'mode': 'F', 'timeindex': [16, 156, 318, 365, 0, 0, 403, 0], 'flavors': [5.0, 5.0, 5.0, 5.0, 5.0, 5.0, 5.0, 5.0, 5.0], 'flavorlabels': ['Acidity', 'Aftertaste', 'Clean Cup', 'Head', 'Fragrance', 'Sweetness', 'Aroma', 'Balance', 'Body'], 'flavorstartangle': 90, 'flavoraspect': 1.0, 'title': 'Guatemala 800', 'beans': '', 'weight': [800.0, 0.0, 'g'], 'volume': [0.0, 0.0, 'ml'], 'density': [0.0, 'g', 1, 'l'], 'density_roasted': [0.0, 'g', 1, 'l'], 'roastertype': 'Arc 800', 'machinesetup': '800', 'operator': 'T Morsanutto', 'organization': '', 'drumspeed': '55', 'heavyFC': False, 'lowFC': False, 'lightCut': False, 'darkCut': False, 'drops': False, 'oily': False, 'uneven': False, 'tipping': False, 'scorching': False, 'divots': False, 'whole_color': 0, 'ground_color': 0, 'color_system': '', 'volumeCalcWeightIn': '', 'volumeCalcWeightOut': '', 'roastdate': 'Sun Sep 13 2020', 'roastisodate': '2020-09-13', 'roasttime': '14:21:19', 'roastepoch': 1600021279, 'roasttzoffset': 18000, 'roastbatchnr': 0, 'roastbatchprefix': '', 'roastbatchpos': 1, 'roastUUID': 'ddce9580b8db4f818717a9e47a6a0684', 'beansize': '0.0', 'beansize_min': '0', 'beansize_max': '0', 'specialevents': [], 'specialeventstype': [], 'specialeventsvalue': [], 'specialeventsStrings': [], 'etypes': ['Air', 'Drum', 'Damper', 'Burner', '--'], 'roastingnotes': '', 'cuppingnotes': '', 'timex': [0.1870376815, 2.137801817, 4.1590230475, 6.148950039, 8.1381813865, 10.1550700965, 12.181959901, 14.1421160305, 16.161316028, 18.18241982, 20.1429561805, 22.1585042055, 24.148506091, 26.1407053765, 28.158264205, 30.1484608445, 32.1415084435, 34.157849042, 36.150265119, 38.138309235, 40.161293893, 42.1770346785, 44.169068636, 46.162114379, 48.146971622, 50.1381512625, 52.1594021435, 54.187459352, 56.137315537, 58.217222861, 60.17814326, 62.137949075, 64.157460837, 66.149904917, 68.138167577, 70.1578763785, 72.184259918, 74.1386149215, 76.187074625, 78.149520889, 80.168784044, 82.161315914, 84.148114929, 86.168760429, 88.1563764375, 90.1569802615, 92.167497708, 94.1562767, 96.147655624, 98.1378782015, 100.1568506635, 102.148844607, 104.1685673035, 106.1653630385, 108.1784314735, 110.1672064955, 112.1565943195, 114.1465844535, 116.1364159505, 118.15611071, 120.148244104, 122.1375861, 124.159035791, 126.136290598, 128.1700931675, 130.156846227, 132.1463539105, 134.1655867525, 136.1575640355, 138.156744356, 140.1684348865, 142.157210305, 144.1460528925, 146.1675715715, 148.155466436, 150.14592698, 152.1381219395, 154.16870137, 156.175594751, 158.1664454405, 160.1440501885, 162.147681834, 164.165750163, 166.1582380395, 168.145632631, 170.1743460615, 172.1555804, 174.1470637195, 176.169422839, 178.156487235, 180.1462861085, 182.1659540775, 184.155468781, 186.155264388, 188.164757945, 190.1265411345, 192.145845357, 194.135385038, 196.155390512, 198.144976337, 200.1357206075, 202.16501262, 204.1750256915, 206.166199167, 208.1555221045, 210.147854718, 212.167395517, 214.155499062, 216.1454398345, 218.1719275855, 220.155414441, 222.175278458, 224.1347121495, 226.1428676535, 228.1446257505, 230.135169441, 232.1553449865, 234.145837042, 236.164578695, 238.1551535985, 240.145305098, 242.1644069845, 244.15493052, 246.1445119085, 248.1637006075, 250.134157031, 252.1750507185, 254.135537227, 256.1604794085, 258.146123911, 260.1665841875, 262.1541768185, 264.1447207185, 266.172159397, 268.154289573, 270.145294077, 272.169249264, 274.155458126, 276.1401676605, 278.1650865885, 280.153672278, 282.1531438955, 284.1636738995, 286.154091179, 288.1437669995, 290.163787232, 292.153356069, 294.1468401575, 296.1636280895, 298.163507431, 300.1734965675, 302.158394671, 304.1515582705, 306.143988875, 308.163883872, 310.1533141485, 312.144378239, 316.1550862135, 318.143261268, 320.1669830325, 322.1533278745, 324.1443528305, 326.1634927715, 328.153211345, 330.154969176, 332.163702125, 334.153620874, 336.13288665, 338.1628875105, 340.1536476585, 342.1427543485, 344.163477392, 346.1656264935, 348.172742352, 350.162454811, 352.152077185, 354.1426469085, 356.16409952, 358.153902175, 360.142063061, 362.133914833, 364.153014928, 366.1445296675, 368.1631648315, 370.153003248, 372.142333984, 374.1648394655, 376.151957497, 378.154236998, 380.1629470495, 382.153418837, 384.141674014, 386.1619555675, 388.151453246, 390.142079397, 392.161976835, 394.1627900615, 396.172919544, 398.163698709, 400.154402781, 402.1415847895, 404.1619371585, 406.1541978195, 408.141825567, 410.171690405, 412.1548248575, 414.142278469, 416.1616593615, 418.152852231, 420.141623513, 422.1621537225, 424.1533269215, 426.151570319, 428.1646370325, 430.152730637, 432.1282975755, 434.162508607, 436.1526049235, 438.1323782165, 440.163933053, 442.160792583, 444.1727204885, 446.1315862225, 448.1517790425, 450.141604702, 452.1607805915, 454.1508723035, 456.141090952, 458.173446711, 460.151562876, 462.165225166, 464.161408457, 466.152011523, 468.143116521, 470.160349376, 472.151942433, 474.1500939635, 476.1626127745, 478.1515958295, 480.1400226505, 482.1572179825, 484.1504815255, 486.1410405445, 488.160445422, 490.161747753, 492.172864381, 494.160359436, 496.153474558, 498.1403080095, 500.1603684565, 502.1497510765, 504.1402927545, 506.1695135955, 508.1515036635, 510.1349468895, 512.161990485, 514.149910347, 516.1398137845, 518.130258926, 520.15082 [... cut ...] ightout': 0.0, 'ambient_temperature': 104.7, 'det': 15.355730209935093, 'dbt': 19.758385605748536}, 'anno_positions': [[0, 32.14150844349979, 483.96, 32.14150844349979, 383.96], [-1, 120.14824410400001, 258.24, 120.14824410400001, 158.24], [1, 312.14437823900016, 347.14, 312.14437823900016, 247.14], [2, 638.1672513510002, 451.54, 638.1672513510002, 351.54], [3, 732.1665298344999, 481.95, 732.1665298344999, 320.95], [6, 808.1427728845003, 512.26, 808.1427728845003, 374.26]], 'flag_positions': []}

roasterracer, 2020-09-13:
here's another

{'version': '2.4.1', 'revision': 'd6d5229', 'build': '0', 'artisan_os': 'Mac OS X', 'artisan_os_version': '10.14.6', 'mode': 'F', 'timeindex': [13, 147, 270, 319, 0, 0, 394, 0], 'flavors': [5.0, 5.0, 5.0, 5.0, 5.0, 5.0, 5.0, 5.0, 5.0], 'flavorlabels': ['Acidity', 'Aftertaste', 'Clean Cup', 'Head', 'Fragrance', 'Sweetness', 'Aroma', 'Balance', 'Body'], 'flavorstartangle': 90, 'flavoraspect': 1.0, 'title': 'Brazil 800', 'beans': '', 'weight': [800.0, 0, 'g'], 'volume': [0.0, 0, 'ml'], 'density': [0.0, 'g', 1, 'l'], 'density_roasted': [0.0, 'g', 1, 'l'], 'roastertype': 'Arc 800', 'machinesetup': '800', 'operator': 'T Morsanutto', 'organization': '', 'drumspeed': '55', 'heavyFC': False, 'lowFC': False, 'lightCut': False, 'darkCut': False, 'drops': False, 'oily': False, 'uneven': False, 'tipping': False, 'scorching': False, 'divots': False, 'whole_color': 0, 'ground_color': 0, 'color_system': '', 'volumeCalcWeightIn': '', 'volumeCalcWeightOut': '', 'roastdate': 'Sun Sep 13 2020', 'roastisodate': '2020-09-13', 'roasttime': '14:42:53', 'roastepoch': 1600022573, 'roasttzoffset': 18000, 'roastbatchnr': 0, 'roastbatchprefix': '', 'roastbatchpos': 1, 'roastUUID': '1e263249c5a543c7b0d04a03431c4384', 'beansize': '0.0', 'beansize_min': '0', 'beansize_max': '0', 'specialevents': [], 'specialeventstype': [], 'specialeventsvalue': [], 'specialeventsStrings': [], 'etypes': ['Air', 'Drum', 'Damper', 'Burner', '--'], 'roastingnotes': '', 'cuppingnotes': '', 'timex': [0.594904242, 2.580543314, 4.600863927, 6.591195672, 8.583850246, 10.604131762, 12.600385804, 14.5808686845, 16.572846228, 18.5913588485, 20.581430655, 22.572624813, 24.5943348805, 26.5812754325, 28.6035975995, 30.6204782015, 32.580475966, 34.6017623235, 36.590405841, 38.581643205, 40.571703322, 42.5953777285, 44.59039641, 46.601322346, 48.59081805, 50.580291113, 52.5702676305, 54.5926145935, 56.5801776865, 58.571958198, 60.5976018885, 62.5830538925, 64.601310225, 66.591163789, 68.5811834, 70.599859445, 72.592391096, 74.579665133, 76.6046456395, 78.589845461, 80.5800821305, 82.601377234, 84.5920619275, 86.5795247105, 88.601826432, 90.5683331665, 92.5921054585, 94.6013694985, 96.590174364, 98.5807894645, 100.6000007635, 102.591972994, 104.581941692, 106.6006281095, 108.5999097725, 110.5787502685, 112.601122061, 114.589725466, 116.5823367655, 118.60072821, 120.5910646385, 122.58267978, 124.6041397685, 126.5913987675, 128.5802779845, 130.6007630145, 132.591375576, 134.5794287535, 136.5997173015, 138.5888304135, 140.5907651005, 142.5999784525, 144.5899229525, 146.5817633565, 148.5999304075, 150.5916917965, 152.5798118925, 154.6002383725, 156.595891477, 158.5790278605, 160.600967885, 162.5678028025, 164.5785005375, 166.599047848, 168.5883551665, 170.5814248325, 172.603280998, 174.618288638, 176.5800658825, 178.598834951, 180.574451789, 182.5799922885, 184.6006029165, 186.590527439, 188.58681941, 190.5722871865, 192.588907928, 194.57900038, 196.5749534905, 198.5915358645, 200.5816505545, 202.6002686145, 204.5959406855, 206.5796834585, 208.597757798, 210.5876544945, 212.577564851, 214.5988524645, 216.58818197, 218.580660687, 220.628689058, 222.58994269, 224.5804691275, 226.599240556, 228.5880447715, 230.5775402455, 232.597109233, 234.590263319, 236.5858253295, 238.6009942375, 240.5885377375, 242.577145569, 244.5677173875, 246.5877975325, 248.578929568, 250.5971130305, 252.594491081, 254.577577259, 256.597559992, 258.58689992, 260.5769061165, 262.597391284, 264.5866185945, 266.5780707695, 268.601892119, 270.616624563, 272.578510436, 274.5988649645, 276.5867164845, 278.5772200075, 280.597866532, 282.5871259985, 284.586465843, 286.59964544, 288.590896094, 290.577089332, 292.5969789665, 294.58636741, 298.5981543645, 300.595007128, 302.576779826, 304.5957532535, 306.5855585405, 308.5768102255, 310.596074169, 312.586665654, 314.5773606855, 316.626534236, 318.6165990875, 320.5778983015, 322.596678344, 324.586112969, 326.577188775, 328.5971867815, 330.5895648405, 332.6055714285, 334.5694444015, 336.5876247285, 338.578751833, 340.5981857365, 342.586740186, 344.577720539, 346.5954600865, 348.5929970395, 350.576172769, 352.5985241285, 354.5865961715, 356.5772262655, 358.595369882, 360.5873150325, 362.574778445, 364.5967288475, 366.6153591955, 368.5783456315, 370.5949358335, 372.587051263, 374.5762303805, 376.5970137695, 378.5875400025, 380.5848900915, 382.5956260865, 384.585459005, 386.577444097, 388.5948697795, 390.586743117, 392.577342645, 394.5978493975, 396.5922659155, 398.5747107275, 400.5954460555, 402.5845323925, 404.574709321, 406.595170652, 408.5845157605, 410.575659029, 412.623322094, 414.5868750715, 416.5787763495, 418.59626159, 420.5868179365, 422.5750198975, 424.5954389765, 426.5842576755, 428.5830453215, 430.595458491, 432.5841104805, 434.574095724, 436.593847581, 438.5841294555, 440.574086089, 442.5958978775, 444.59393698, 446.5757020055, 448.5964372305, 450.58378405, 452.574494182, 454.5944317525, 456.584387411, 458.5737991, 460.595508801, 462.6142800055, 464.573736295, 466.593670992, 468.583521868, 470.5741478915, 472.565454012, 474.583711395, 476.5822303975, 478.593404004, 480.585573978, 482.579019294, 484.595246786, 486.596464699, 488.573910102, 490.56753463, 492.5937316855, 494.575021784, 496.5936925265, 498.5837626935, 500.5745109485, 502.5944698305, 504.584921394, 506.5750362265, 508.5954903155, 510.5834897935, 512.5725910485, 514.5961914255, 516.578004404, 518.5732193815, 520.570357498, 522.583284559, 524.608291587, 526.6239993575, 52 [... cut ...]  0, 'ambient_temperature': 105.8, 'det': 9.764423700573703, 'dbt': 22.760143512899663}, 'anno_positions': [[0, 26.58127543249975, 488.77, 26.58127543249975, 388.77], [-1, 112.60112206099939, 257.95, 112.60112206099939, 157.95], [1, 294.58636740999964, 350.67, 294.58636740999964, 250.67000000000002], [2, 542.6093166824994, 442.83, 542.6093166824994, 342.83], [3, 640.5916738650003, 473.74, 640.5916738650003, 312.74], [6, 790.5879307214996, 503.82, 790.5879307214996, 381.82]], 'flag_positions': []}

MAKOMO, 2020-09-14:
Yes, same result as before. Could you verify if the new duplicate filter was active (menu Config >> Curve, 2nd tab "Filters"?

[image]

MAKOMO, 2020-09-14:
It seems to me from those readings that this machine is not delivering new readings faster than 4sec. So you should set your sampling rate down from 2sec to 4sec and ensure that you have oversampling deactivated.

closed 2020-09-16 by MAKOMO

roasterracer, 2020-09-18:
sorry it took me a while to respond. i usually roast every few days. anyways turned out i used an earlier 2.4.1 build that didn't have that filter you screenshot. i redownloaded the continuous build and was able to apply the filter, and made the adjustments you suggested with sampling. I think it worked out well. see attached. 

much appreciated

[Colombia 400_20-09-17_2259 - 427 quick good  copy.txt](https://github.com/artisan-roaster-scope/artisan/files/5243488/Colombia.400_20-09-17_2259.-.427.quick.good.copy.txt)

MAKOMO, 2020-09-18:
Yes, here the interpolating filter worked. Thanks for this further test. Still this is an amazingly bad signal at 4sec sampling interval. I wonder what this machine is doing internally.

roasterracer, 2020-09-21:
lol when i fill it with green coffee the machine roasts them (with a little
help from me of course)  ;)

thanks again Marko!