artisan-roaster-scope/artisan #493: Curves Fonts Problem

wuyi-ligux, 2020-07-24:
Windows：Curves text is not supported Traditional Chinese
Linux：Curves text is not supported Traditional Chinese and Simplified Chinese

Change font doesn't work
[image]

wuyi-ligux, 2020-07-24:
Can add open source fonts：wqy？ http://wenq.org/ https://sourceforge.net/projects/wqy/files/

MAKOMO, 2020-07-26:
Added this font and also the open-source Adobe "Source Han Serif" font in all its language specific variants. Could you give this a quick test and report if this works as expected?

Also I wonder if you would be possible for you to update the traditional Chinese translations in artisan_zh_TW.ts along the lines you did in artisan_zh_CN.ts? Note that you can edit those .ts files using the Qt Linguist app that you  can download. The .qm files are generated automatically from the .ts files via the Artisan build scripts.

wuyi-ligux, 2020-07-27:
OK, let me test it first. I have finished the translation file of TW, but like CN, some words need to be optimized and modified, and then I submit them together.

wuyi-ligux, 2020-07-27:
Wonderful！All fonts can work，wen quan yi no need to distinguish between simplified and traditional Chinese！

wuyi-levard, 2020-07-28:
> Added this font and also the open-source Adobe "Source Han Serif" font in all its language specific variants. Could you give this a quick test and report if this works as expected?
>
> Also I wonder if [...]

Translation completed！ Pull requests: update Chinese translations(CN and TW) #494

closed 2020-07-30 by wuyi-levard