artisan-roaster-scope/artisan #495: 2.4 not working on Mac with Hottop 2K+

awer25, 2020-08-09:
## Expected Behavior
Version 2.4 would read temps and control my Hottop KN-8828B-2K+.

## Actual Behavior
Version 2.4 does not read temps or control my Hottop despite choosing the correct Comm Port. Version 2.1.2 does work properly, however. Also, 2.4 did work on a Windows laptop with this roaster.

## Steps to Reproduce the Problem

  1. Plug in Hottop KN-8828B-2K+ and turn on to Auto
  2. Install Artisan 2.4
  3. Config->Machine->Hottop->KN-8828B-2K+
  4. Choose Comm Port "FT230X Basic UART"
  5. Press "On".

Result: I believe temps read "-0.1" or similar. No control over unit or signs of communication between the unit and the computer. 

Downgrading to 2.1.2 works perfectly. Re-upgrading to 2.4 breaks the system again.

## Specifications

  - Artisan Version: 2.4
  - Artisan Build (number in brackets shown in the about box): No longer installed, but it was the most recent one available on August 9, 2020.
  - Platform (Mac/Windows/Linux + OS version): MacOS 10.15.6 (non-beta)
  - Connected devices or roasting machine: 2017 15" Macbook Pro via Apple USB-C to USB-A adapter.
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles. I no longer have 2.4 installed, but I used the default settings.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2020-08-10:
Thanks for your report, but this is a duplicate (see Issue #483). Either download a current development build that fixes this or use version 2.1.2 until the next official release is available

closed 2020-08-10 by MAKOMO

awer25, 2020-08-10:
Thank you, and sorry for the duplicate.